balena-os/meta-balena #1182: [aarch64] os-config: undefined instruction

telphan, 2018-09-10:
I do not have a way to consistently reproduce this, it shows up randomly. I have observed this on aarch64, but might not be limited to this architecture. 
```
[   10.620694] os-config[959]: undefined instruction: pc=00000000003ed2e4
[   10.627381] Code: aa1e03e2 b90013e3 910043e3 d61f01e0 (f81f0ffe) 
         Starting Create Static Device Nodes in /dev...
[FAILED] Failed to start OS configuration device key service.
See 'systemctl status os-config-devicekey.service' for details.
[DEPEND] Dependency failed for Prepare OpenVPN.
[DEPEND] Dependency failed for OpenVPN.
[DEPEND] Dependency failed for Resin supervisor.
```

agherzan, 2018-09-10:
Thanks @telphan . Feel free to update this issue whenever you have updates on this issue.

agherzan, 2018-10-19:
@telphan what is the status of this?

ZubairLK, 2020-01-27:
Closing stale issues for cleanup. Can reopen if needed

closed 2020-01-27 by ZubairLK