balena-os/meta-balena #1194: Balena Build Issue

jsrc27, 2018-09-19:
I'm currently working with an openembedded based image with container support. Due to storage constraints I'm now aiming to use balena in place of docker in my image. I copied the balena recipe and it's dependencies into my own layer, but when building I run into the below issue:

Log data follows:
| DEBUG: Executing shell function do_compile
| # WARNING! I don't seem to be running in a Docker container.
| # The result of this command might be an incorrect build, and will not be
| # officially supported.
| #
| # Try this instead: make all
| #
| 
| Package devmapper was not found in the pkg-config search path.
| Perhaps you should add the directory containing `devmapper.pc'
| to the PKG_CONFIG_PATH environment variable
| No package 'devmapper' found
| Removing bundles/
| 
| ---> Making bundle: dynbinary-balena (in bundles/dynbinary-balena)
| Building: bundles/dynbinary-balena/balena-17.12.0-dev
| # github.com/docker/docker/pkg/ioutils
| .gopath/src/github.com/docker/docker/pkg/ioutils/eagerwriter.go:41:3: undefined: syscall.SyncFileRange
| .gopath/src/github.com/docker/docker/pkg/ioutils/eagerwriter.go:48:2: undefined: syscall.SyncFileRange
| WARNING: exit code 1 from a shell command.

I'm not awfully familiar with Go so any advice would be greatly appreciated.

ZubairLK, 2018-09-21:
@jsrc27 Have you had a chance to look at a go patch for arm in meta-resin?

https://github.com/resin-os/meta-resin/blob/ede6ca349df939ba1c223c76c3e378c83f102732/meta-resin-sumo/recipes-devtools/go/go-1.9/0001-Implement-arm_sync_file_range-and-SyncFileRange-wrap.patch

Out of curiosity, do you mind sharing more details about your project?

Thanks
ZubairLK

agherzan, 2018-10-19:
@jsrc27 I will close this issue because that error should be fixed by the patch @ZubairLK referenced above. If that is not the case, tell us and we can investigate further. Thanks.

closed 2018-10-19 by agherzan