balena-os/meta-balena #1199: NetworkManager's connectivity section

majorz, 2018-09-24:
Defining a `connectivity` section in the global NetworkManager's config helps in improving the overall connectivity of the device when a secondary networking device exists.

When the primary networking device looses Internet connectivity, but is still up, then a secondary device with connectivity is not used. This affects devices that use a combination of a primary and a backup connection (WiFi + cellular or Ethernet + WiFi). The fix is simple and requires enabling the Internet connectivity check in our global NetworkManager config: https://developer.gnome.org/NetworkManager/stable/NetworkManager.conf.html

NetworkManager iterates through all the networking devices and uses libcurl to fetch the remote page defined in the `connectivity` section. This is undocumented, but NetworkManager uses these results to lower the route metrics for the interfaces with broken default routes.

After we expose a new URL for this purpose on our website, we need to enable the connectivity check in our OS. https://github.com/resin-io/hq/issues/1474.

ZubairLK, 2018-09-24:
I think we discussed this and then had the concern that it eats data over time..
5 minute ping 24/7 can add up

And the alternative we used to check internet connectivity was to check if the VPN was connected or not..

majorz, 2018-09-24:
The purpose of this is to reestablish connectivity through a secondary networking device when the primary one fails to provide it.

shaunmulligan, 2018-09-24:
Perhaps we can have an option to disable this check like we have for the others, so people sensitive to bandwidth usage can disable it, but its on by default. I agree with @majorz that this will be useful.

It we have it on by default though, it does cause an issue when people update to this version and don't know its been implemented.

shaunmulligan, 2018-09-24:
We could also actually allow the user to set the `interval` option to any `x` number of seconds they wish

majorz, 2018-09-24:
Right, we need to have that configurable when we allow modifying the global NetworkManager config in the future. There are other useful options as well.

majorz, 2018-09-24:
Alternatively we can do a more robust implementation on our own: that is to do what NetworkManager does with penalizing interfaces with default routes that do not provide Internet connectivity, but doing all of that only when the VPN is down. This should solve the bandwidth issue. We will still need the connectivity check URL. To shorten the implementation path we can use the NetworkManager's API - that is to enable/trigger the connectivity check only when the VPN is down. Something along those lines will solve the bandwidth concern. If we go such a road, we need to ensure we do not break connectivity in attempt to improve it.

majorz, 2018-10-12:
This should be a nice fit for our upcoming higher-level networking service (the one that will contain WiFi Connect's functionality).

agherzan, 2018-11-02:
Discussed in 2018-11-02
The solution is to open the ability for people to set a configuration in the config.json and the OS will translate at boot that to a /etc/NetworkManager/conf.d

@majorz this should enable overwriting any default configuration we currently provide.

majorz, 2019-01-23:
The calculated packet sizes are quite small. The connectivity recalculation is triggered when connection state changes, so we can safely do an interval of one hour as we discussed. We can enable this as a non-configurable setting when the API endpoint is ready.

pcarranzav, 2019-03-28:
Discussed in architecture call today:

* The default behavior should be to use the currently proposed approach (the API's connectivity-check endpoint, using the API URL that comes from config.json which we use to know what backend we're provisioned on)
* We can have a field in config.json to change the connectivity URL for users that need it.

PR #1272 (2018-11-14): Add os-networkmanagerconf and fix custom udev rules cleanup
Fixes #1199 

This PR allows custom config fragments to be passed to NetworkManager via config.json.
Also fixes custom udev rules so that it cleans up correctly if rules are removed from config.json

To turn a config fragment into the format that can be easily added to config.json, use
 `cat conf_filename | jq -sR .`
e.g.
```
root@balena:~# cat /etc/NetworkManager/conf.d/20.conf | jq -sR .
"[connectivity]\nuri=http://www.archlinux.org/check_network_status.txt\ninterval=10\n"
root@balena:~# cat /etc/NetworkManager/conf.d/21.conf | jq - [... cut]
PR #1468 (2019-05-10): Add support to update a connectivity section in NetworkManager via config.json
We don't have a connectivity section configured in NetworkManager config at the moment.
This causes issues for devices with multiple interfaces that are active but only one has actual internet connectivity. NM doesn't know and does not adjust route metrics.

We now default to having the following in NetworkManager.conf

```
[connectivity]
uri=https://api.balena-cloud.com/connectivity-check
interval=3600
request=
```

The uri and interval are editable via config.json. See readme.md

Fixes #1199 

---
### Contributor  [... cut]
closed 2019-12-02 by ZubairLK