balena-os/meta-balena #1203: How to specify mount options

svenschwermer, 2018-09-27:
Hi,

In current versions of meta-resin, how does one best supply mount options (e.g. `sync`)? I'm particularly interested in doing that for the state and data partitions.

@agherzan I believe you built the `resin-mounts` recipe and might be the best to answer this :-)

Thanks, Sven

ZubairLK, 2018-09-27:
sync might have a performance penalty if you are doing lots of writes. Can you try the following to see if the following works for you?

Inside a container
`mount -o remount,sync /data`

In the host os
`mount -o remount,sync /resin-data`
`mount -o remount,sync /mnt/state`

ZubairLK, 2018-09-27:
The one inside the container should be possible to do via the dockerfile itself.

ZubairLK, 2018-09-27:
For the state mount, the following service triggers the mount.
https://github.com/resin-os/meta-resin/blob/master/meta-resin-common/recipes-support/resin-mounts/resin-mounts/resin-state.service

The actual file that does the mounting is 
https://github.com/resin-os/meta-resin/blob/master/meta-resin-common/recipes-support/resin-mounts/resin-mounts/resin-partition-mounter#L105

That line will need an extra option. Something like `-o sync`

svenschwermer, 2018-09-28:
I had seen that `resin-partition-mounter` does the actual mounting. I was hoping that there would be a canonical way to specify mount options.

svenschwermer, 2018-09-28:
> sync might have a performance penalty if you are doing lots of writes. Can you try the following to see if the following works for you?
> 
> Inside a container
> `mount -o remount,sync /data`
> 
> In the host os
> `mount -o remount,sync /resin-data`
> `mount -o remount,sync /mnt/state`

I have succeeded to mount the data partition `sync`. However, I can't get the state partition to be mounted `sync`. I tried both in Resin OS 2.9.7 (by modifying the `mnt-{state,data}.mount` systemd unit file) and 2.19.0 (by modifying the `resin-partition-mounter` helper script).

In the 2.19.0 journal, I can see that the state partition is already mounted when the helper script is called—why's that?
`Sep 28 11:18:23 localhost resin-partition-mounter[441]: INFO: /dev/disk/by-label/resin-state already mounted in /mnt/state.`

ZubairLK, 2018-09-28:
@svenschwermer Looks like the state partition is mounted in the initramfs here 
https://github.com/resin-os/meta-resin/blob/master/meta-resin-common/recipes-core/initrdscripts/files/machineid#L22

ZubairLK, 2018-09-28:
The initramfs can be tricky to debug. If you have uart, add `shell` in the kernel cmdline and boot will stop during initramfs. Then `exit` again and again to run the scripts in `/init.d/` folder inside the initramfs