balena-os/meta-balena #1205: OpenVPN disconnect after establishing SSH session

majorz, 2018-09-27:
A RPi3 B+ device running for a couple of days left on the desk with Ethernet connection only. I opened the dashboard and established an SSH session, a couple of minutes later OpenVPN disconnected and reconnected. There were not any previous OpenVPN disconnects in the logs from time the device was started, which lead me to think that my SSHing into the device and the OpenVPN disconnect are related somehow.

Logs here: https://gist.github.com/majorz/be3a7a46b25ad3372f803465f338a224

agherzan, 2018-10-01:
@majorz Do you have a consistent way to reproduce this?

majorz, 2018-10-03:
@agherzan I will leave the RPi3 B+ running for a longer time to see whether this can be reproduced

ZubairLK, 2020-01-27:
Closing issue to cleanup. Can reopen if still relevant/needed

closed 2020-01-27 by ZubairLK