balena-os/meta-balena #1208: balenaEngine takes a minute to load on intel-nuc on a clean boot

ZubairLK, 2018-10-02:
Have seen this far too many times.

```
root@resin:~# journalctl -u balena
-- Logs begin at Tue 2018-10-02 16:32:49 UTC, end at Tue 2018-10-02 16:34:25 UTC. --
Oct 02 16:32:50 resin systemd[1]: Starting Balena Application Container Engine...
Oct 02 16:33:54 resin healthdog[566]: time="2018-10-02T16:33:54Z" level=warning msg="Runnin>
Oct 02 16:33:54 resin healthdog[566]: time="2018-10-02T16:33:54.967484335Z" level=warning m>
Oct 02 16:33:54 resin healthdog[566]: time="2018-10-02T16:33:54.969848143Z" level=info msg=>
```

On a VM running the nuc image running v2.19 and some of my PRs..

ZubairLK, 2018-10-19:
Its a kernel patch that changed the way blocking/non-blocking access to random happened..
More info in https://github.com/resin-os/balena-engine/issues/105 and https://github.com/resin-os/resin-intel/pull/160

robertgzr, 2018-12-11:
@agherzan @ZubairLK 

Once https://github.com/balena-os/balena-engine/pull/123 is merged, can we produce a nuc image with the fix?

agherzan, 2018-12-11:
@robertgzr Ping/create an issue when we merge that PR so we can include it in the OS. The jenkins jobs will automatically provide you with testing images.

robertgzr, 2018-12-18:
@agherzan the fixes are merged into the `17.12-resin` branch

PR #1336 (2018-12-18): balena: Update to 17.12-resin's HEAD
This includes changes which target fixing hanging on low entropy
situation.

Fixes #1208

Change-type: patch
Changelog-entry: Update balena-engine to include low entropy fixes

---
### Contributor checklist

- [ ] Changes have been tested
- [x] `Change-type` present on at least one commit
- [x] `Signed-off-by` is present
- [x] The PR complies with the [Open Embedded Commit Patch Message Guidelines](http://www.openembedded.org/wiki/Commit_Patch_Message_Guidelines)

### Reviewer Guidelines
- When submitting a review, please pick:
  - '*Approve*' if this chang [... cut]
commit 32726b55 (2018-12-18): balena: Update to 17.12-resin's HEAD

This includes changes which target fixing hanging on low entropy
situation.

Fixes #1208

Change-type: patch
Changelog-entry: Update balena-engine to include low entropy fixes
closed 2018-12-18 by agherzan