balena-os/meta-balena #1225: Unreliable udev behaviour in containers with hotplugged storage devices

CameronDiver, 2018-10-15:
Running a privileged container with `UDEV=on` and hotplugging a storage device does not result in the device files being created.

Without `UDEV=on`, the device files aren't created at all.

With `UDEV=on, the "root" device file (e.g. `/dev/sda`) is created, but any partition device files are not (e.g. `/dev/sda1`)

This is reproducable at least on an rpi3.

memory, 2018-10-16:
Actually I may have spoken too soon there: 2.13.6r1's behavior seems to be...inconsistent. :(

memory, 2018-10-17:
another data point: udev events / usb devices _do_ appear to still work as expected if and only if I run with `ENV INITSYSTEM on` inside the container.

shaunmulligan, 2018-12-05:
@memory and @CameronDiver is there a simple reproduction of this issue? I want to try test this against our new base images, e.g.: `balenalib/raspberrypi3-debian` since there we have udevd running in a network namespace, which seems to have improved things across the board for other udev related issues. So a simple test case would be great to run on those images.