balena-os/meta-balena #1247: Device loses time sync after application update

xginn8, 2018-10-22:
The error logged is `Could not resolve address of initstepslew server 0.resinio.pool.ntp.org`. Disabling udev seems to fix the issue. Seems like a funky chrony + udev interaction, since the device uses iMX7 and its driver creates a HW RTC node.

xginn8, 2018-10-22:
cc @ZubairLK

agherzan, 2018-10-22:
@xginn8 Do you have a way to reproduce?

agherzan, 2018-11-02:
@xginn8 It would really help to have a way to reproduce. The error posted there doesn't seem to show anything related to rtc. It seems it was just not yet connected to the network.

ZubairLK, 2018-12-04:
Traced it down to a privileged udev container.

```
zubairlk@zubair-xps-resin:~$ cat docker-compose.yml 
version: '2.1'
volumes:
  resin-data: {}
services:
  test-chrony:
    build: .
    restart: always
    privileged: true
    command: sleep 10000
zubairlk@zubair-xps-resin:~$ cat Dockerfile.template 
FROM resin/%%RESIN_MACHINE_NAME%%-debian
# we need to receive udev events from the host
ENV UDEV on
CMD ["sleep","100000"]
zubairlk@zubair-xps-resin:~/Downloads/testpichrony/multicontainer-getting-started$ 
```

Run `while true; do timedatectl | grep -i synch; sleep 0.1; done` in Host OS.
Then restart container via dashboard. (or update)

Its not a chrony or a device issue. As I see this in 2.12.7 on the NUC as well.
timesyncd just gets back in sync in <0.5 seconds. Hence it went by unnoticed.
chronyd can be sleeping while an application update/restart happens which highlights the issue severely.

Also note: clicking the restart button on the dashboard (or update) destroys and recreates a container. That causes it. If you click stop/start buttons to stop/start the container. Time sync won't be lost.

My theory is that: during initialisation something tells the kernel and just clears the time sync status by sending `STA_UNSYNC`. Not sure where to look though..

cc @agherzan

shaunmulligan, 2018-12-04:
@ZubairLK can you do the same test in a `balenalib/%%BALENA_MACHINE_NAME%%-debian` image, in those images we have now started running the container udev in a network namespace and it has solved the issue for modems getting unplugged in the kernel at startup, so i wonder if it will solve this too.

ZubairLK, 2018-12-06:
@shaunmulligan changing to `balenalib/%%BALENA_MACHINE_NAME%%-debian` didn't change anything and time still looses sync on the app container restart above

ZubairLK, 2019-01-22:
This was traced to a udev rule inside the container.
Depends on the base image. For debian, e.g

```
root@af6221a:/# cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 8 (jessie)"
NAME="Debian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=debian
HOME_URL="http://www.debian.org/"
SUPPORT_URL="http://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
root@af6221a:/#
root@af6221a:/# cat ./lib/udev/rules.d/85-hwclock.rules
# Set the System Time from the Hardware Clock and set the kernel's timezone
# value to the local timezone when the kernel clock module is loaded.

KERNEL=="rtc0", RUN+="/lib/udev/hwclock-set $root/$name"
```

When a container is privileged and UDEV is enabled, this rule messed up the hwclock time sync. And it takes a while for chrony to sync again depending if the daemon is sleeping etc.

This explains why an application container update or (restart via dashboard which destroys/recreates the container) would trigger this issue.
Just stopping/starting the container would not cause it.

Closing this issue as there isn't much we can do except suggest that if you have this issue, find the udev rule causing it and add an `rm` in your Dockerfile for that rule.