balena-os/meta-balena #1254: systemd-tmpfiles-clean fails

svenschwermer, 2018-10-29:
Since there is a symlink `/var/tmp -> /var/volatile/tmp`, systemd-tmpfiles-clean fails with `systemd-tmpfiles[1886]: Failed to open directory /var/tmp: Too many levels of symbolic links`. This is this issue: https://bugzilla.redhat.com/show_bug.cgi?id=1334881 That means that temporary files will never be cleaned up.

agherzan, 2018-10-30:
Hi @svenschwermer I just tried to reproduce this on a latest generic x86 image and it seems like the service runs just fine:

```
root@balena:~# systemctl status systemd-tmpfiles-clean
● systemd-tmpfiles-clean.service - Cleanup of Temporary Directories
   Loaded: loaded (/lib/systemd/system/systemd-tmpfiles-clean.service; static; vendor preset: enabled)
   Active: inactive (dead) since Tue 2018-10-30 16:47:53 UTC; 9s ago
     Docs: man:tmpfiles.d(5)
           man:systemd-tmpfiles(8)
  Process: 1143 ExecStart=/bin/systemd-tmpfiles --clean (code=exited, status=0/SUCCESS)
 Main PID: 1143 (code=exited, status=0/SUCCESS)

Oct 30 16:47:53 balena systemd[1]: Starting Cleanup of Temporary Directories...
Oct 30 16:47:53 balena systemd[1]: Started Cleanup of Temporary Directories.
```

I've tried this on `2.24.0+rev3`, the latest version available in production. Could you try to reproduce on that as well?

svenschwermer, 2018-10-30:
Can you check whether `/var/tmp` is a symlink on your test system?

agherzan, 2018-10-30:
@svenschwermer yes it is. This is part of the fs setup we have for volatile:

`lrwxrwxrwx  1 root root   12 Oct 30 16:24 tmp -> volatile/tmp`

agherzan, 2018-10-30:
`lrwxrwxrwx  1 root root   12 Oct 30 16:24 tmp -> volatile/tmp`

svenschwermer, 2018-10-31:
Weird. This is what I have:
```sh
root@839e392:~# journalctl --unit=systemd-tmpfiles-clean
-- Logs begin at Wed 2018-10-31 16:41:54 UTC, end at Wed 2018-10-31 17:42:28 UTC. --
Oct 31 16:56:48 839e392 systemd[1]: Starting Cleanup of Temporary Directories...
Oct 31 16:56:48 839e392 systemd-tmpfiles[1844]: Failed to open directory /var/tmp: Too many levels of symbolic links
Oct 31 16:56:48 839e392 systemd[1]: Started Cleanup of Temporary Directories.

root@839e392:~# systemctl status systemd-tmpfiles-clean
● systemd-tmpfiles-clean.service - Cleanup of Temporary Directories
   Loaded: loaded (/lib/systemd/system/systemd-tmpfiles-clean.service; static; vendor preset: enabled)
   Active: inactive (dead) since Wed 2018-10-31 16:56:48 UTC; 48min ago
     Docs: man:tmpfiles.d(5)
           man:systemd-tmpfiles(8)
  Process: 1844 ExecStart=/bin/systemd-tmpfiles --clean (code=exited, status=0/SUCCESS)
 Main PID: 1844 (code=exited, status=0/SUCCESS)

Oct 31 16:56:48 839e392 systemd[1]: Starting Cleanup of Temporary Directories...
Oct 31 16:56:48 839e392 systemd-tmpfiles[1844]: Failed to open directory /var/tmp: Too many levels of symbolic links
Oct 31 16:56:48 839e392 systemd[1]: Started Cleanup of Temporary Directories.

root@839e392:~# grep PRETTY /etc/os-release
PRETTY_NAME="Resin OS 2.23.0+rev1"

root@839e392:~# /lib/systemd/systemd --version
systemd 232
-PAM -AUDIT -SELINUX +IMA -APPARMOR +SMACK +SYSVINIT +UTMP -LIBCRYPTSETUP -GCRYPT -GNUTLS +ACL +XZ -LZ4 -SECCOMP +BLKID -ELFUTILS +KMOD -IDN
```
How do we narrow this down further?

agherzan, 2018-10-31:
What is the version on systemd on that version? `systemctl --version`.

svenschwermer, 2018-10-31:
232, see the last line in my last posting.

-- 
SVEN SCHWERMER <sven.schwermer@disruptive-technologies.com> FIRMWARE
DEVELOPER
[image: DISRUPTIVE TECHNOLOGIES] <http://disruptive-technologies.com/>
7075 TILLER
TRONDHEIM
NORWAY

OFFICE +47 57 98 88 55 <%2B4757988855>
MOBILE +47 94 16 96 42 <%2B4794169642>
LINKEDIN <https://www.linkedin.com/company/disruptive-technologies-as>
The World’s Smallest Wireless Sensor

svenschwermer, 2018-10-31:
> What is the version of systemd on that version? `systemctl --version`.

232, see the last line in my last posting.

agherzan, 2018-10-31:
That is what I suspect. genericx86 images use 237 as they are updated to sumo. Can you test on an updated poky version?

svenschwermer, 2018-11-01:
No, I can't easily. Is this the culprit? https://github.com/systemd/systemd/commit/3d22bc8636b05524a12255781bed6ef5c6afac9d

svenschwermer, 2018-11-01:
@agherzan nope, that wasn't the commit that changed the behaviour.

agherzan, 2018-11-01:
First, we need to first figure out if indeed systemd is the culprit here. So I'd say that testing a new systemd would be the first thing to do. I'm not sure how complicated would that be to manually update systemd in pyro to the version in sumo though.

floion, 2018-11-10:
Hi @svenschwermer , the latest release contains a newer systemd. Can you please check if this issue is still reproducible?

svenschwermer, 2018-11-12:
I think, this is resolved now.

svenschwermer, 2018-11-20:
The issue still exists in 2.26.0+rev1:

```
# journalctl --no-hostname -o short-monotonic --identifier=systemd-tmpfiles
-- Logs begin at Tue 2018-11-20 08:59:38 UTC, end at Tue 2018-11-20 09:17:58 UTC. --
[    8.153121] systemd-tmpfiles[562]: "/var/tmp" already exists and is not a directory.
[    8.153963] systemd-tmpfiles[562]: "/var/log" already exists and is not a directory.
[  900.856140] systemd-tmpfiles[2058]: Failed to open directory /var/tmp: Too many levels of symbolic links
```

agherzan, 2018-11-21:
I managed to get to the bottom of this. It was actually reproducible on my x86 machine too but the error being different was not advertised by the tool. In any way, the behaviour, of tmpfiles not cleaning up that directory was the same. In a little detail, on x86 the error is
```
Cannot open directory "/var/tmp": Not a directory
Directory "/var/tmp": Not a directory
```
It's just that systemd completely ignores this error while in your case it outputs a warning of the entry being a symlink.

Now, `/var/tmp` is indeed a symlink to a `tmpfs` mountpoint `/var/volatile/tmp` which systemd avoids following. We inherit from yocto a volatile tmpfiles entry but sadly, that entry doesn't set an age so it won't clean up the target (`/var/volatile/tmp`) see http://git.yoctoproject.org/cgit/cgit.cgi/poky/tree/meta/recipes-core/systemd/systemd/00-create-volatile.conf . The systemd default tmpfiles configuration for tmp is available [here](https://github.com/systemd/systemd/blob/master/tmpfiles.d/tmp.conf#L12). At this point, it sounds like we need to migrate the systemd provided rules for `/tmp/var` to the yocto provided one and, optionally, avoid the warning by removing the entries from systemd's file.

We could as well modify the systemd file to use the target `/var/volatile/tmp` path but that would not work as the volatile file is parsed first and the second will be ignored:
```
[/usr/lib/tmpfiles.d/tmp.conf:12] Duplicate line for path "/var/volatile/tmp", ignoring.
```
This will need a removal in the yocto volatile file as well so in any way it sounds like we need to correlate a modification in both of these components.