balena-os/meta-balena #1259: resin-vars: Implement custom ssh keys service - not working for me

ashishka, 2018-10-31:
Hello!
I put some ssh keys as array to the os-config.json file, but i no tot see any authorized_keys file in the /home/root/.ssh/ folder when box is booted. Maybe, this keys need to be putted at the top level after first "{" symbol ? or to the other place this json file ?
Do i understood right - we need to put rows like this

"os": {
"sshKeys": [
"ssh-rsa 1stSSHkey",
"ssh-rsa 2ndSSHkey"
]
},

to the os-config.json file and rebuild image without any other changes ?
and this is will work with development=1 or without this too ?

agherzan, 2018-10-31:
Hi @ashishka . Can you please paste here the entire config.json file (after sanitizing it for sensitive information)? As well, after you boot please check and paste the output of `systemctl status os-sshkeys.service` and `ls /home/root/.ssh`.

ashishka, 2018-10-31:
Thank you for answer!

os-config.json look like this

cat ./os-config.json
{
	"services": [
		{
			"id": "openvpn",
			"files": {
				"config": {
					"path": "/etc/openvpn/openvpn.conf",
					"perm": ""
				},
				"ca": {
					"path": "/etc/openvpn/ca.crt",
					"perm": ""
				}
			},
			"systemd_services": ["prepare-openvpn.service", "openvpn.service"]
		},
		{
			"id": "ssh",
			"files": {
				"authorized_keys": {
					"path": "/home/root/.ssh/authorized_keys_remote",
					"perm": "600"
				}
			},
			"systemd_services": []
		}
	],
	"os": {
		"sshKeys": [
			"ssh-rsa AAAAB3NzaC...",
			"ssh-rsa AAAAB3NzaC..."
		]
	},
	"keys": [
		"applicationName",
		"applicationId",
		"deviceType",
		"userId",
		"username",
		"appUpdatePollInterval",
		"listenPort",
		"vpnPort",
		"apiEndpoint",
		"vpnEndpoint",
		"registryEndpoint",
		"deltaEndpoint",
		"pubnubSubscribeKey",
		"pubnubPublishKey",
		"mixpanelToken",
		"apiKey",
		"version",
		"deviceApiKey",
		"registered_at",
		"deviceId"
	],
	"schema_version": "1.0.0"
}

when i see to result /home/root/.ssh i see authorized_keys but this is empty
and result of systemctl status os-sshkeys.service is "No custom SSH public key configured"

agherzan, 2018-10-31:
Right. So the documentation refers to config.json, a file in the boot partition. This file that you point to here is not supposed to be changed and includes a configuration schema for cloud api endpoints that are meant to provide a specific os configuration payload. So, long story short, use config.json in the root of the resin-boot partition to set the ssh keys (https://github.com/balena-os/meta-balena/blob/master/README.md#configjson).

closed 2018-10-31 by agherzan

ashishka, 2018-10-31:
sorry, i tried found config.json inside layers to modify it, can you explain me, where i can found this ?

agherzan, 2018-11-01:
The config.json is initialized in the build as part of the image generation routine when the boot partition is populated (https://github.com/balena-os/meta-balena/blob/master/meta-resin-common/classes/image-resin.bbclass#L196). We don't currently offer a way to deploy/initialize a static one as part of the build (sounds like a feature request) so, currently, the easiest way would be to manually set it by mounting the image after build.

ashishka, 2018-11-01:
Thank you so much!
For now, we did image-resin.bbclass modified and build with changes.

ashishka, 2018-11-01:
Yes, i think so, but i doing first steps in the resin/balena, and will try to increase my experience with it, of course.

agherzan, 2018-11-01:
Got it. Enjoy!

ashishka, 2018-11-01:
Sorry, can you explain one small thing?
I have ssh key added to the config.json and see it inside the /home/root/.ssh/autenticate_keys
But when i boot with this image, i see dropbearkey service not started because it can`t create file /etc/dropbear/dropbear_rsa_host_key - fle system is read only

can i copy this file during build process or we have other way to create or copy this ?

agherzan, 2018-11-01:
It is generated at runtime as /etc/dropbear is a bind mount to rw partition: https://github.com/balena-os/meta-balena/blob/master/meta-resin-common/recipes-support/resin-mounts/resin-mounts.inc#L19