balena-os/meta-balena #1266: RTC gets corrupted

zagatta-sonah, 2018-11-02:
We use balena on a RaspberryPi 3B+ with an pcf8523 RTC (specificly an Arduino based power-board called [SleepyPi](https://spellfoundry.com/product/sleepy-pi-2/)).

The only settings I added to my set-up to get it running is adding a device variable
`RESIN_HOST_CONFIG_dtoverlay = i2c-rtc,pcf8523,pi3-miniuart-bt`
The RTC is especially important, as it is connected to an Arduino that we use for power-control (rebooting, keeping alive, etc) because our devices are not accessible to us, once we ship them.

We can recognize the error by calling `hwclock -r` which then returns either
````
hwclock: Cannot access the Hardware Clock via any known method.
hwclock: Use the --debug option to see the details of our search for an access method.
````

or
````
hwclock: ioctl(RTC_RD_TIME) to /dev/rtc to read the time failed: Input/output error
````

We also found out, that just taking the device from power and then plugging it back in solves the problem temporarily, as the RTC work again after the power cut.

But the RTC's get corrupted within 2-3 days again and we're back at the same problem. 

The suggested solution from the manufacturer of the RTC is to comment out some lines in 
`/lib/udev/hwclock-set` to stop the system from constantly writing on the RTC ([reference](url
https://spellfoundry.com/sleepy-pi/setting-up-the-real-time-clock-on-raspbian-jessie/#Setup_the_RTC)).

I already tried to do that in all containers, that the RTC's get still corrupted.

Another solution ist to deactivate NTP and just activate it manually once a day.
This is currently not possible with balena (I was told by the support) and this is of course only one possible solution that might not even help (as it is a solution for the raspberry pi with a standard raspbian installation).

**I would be even better to get full control of the interaction with the RTC and be able to control when the system is writing to or reading from the RTC.**

One other strange thing I noticed is the response to the suggested [NTP-Check from the Docs  ](url
https://www.balena.io/docs/learn/develop/runtime/#checking-if-device-time-is-ntp-synchronized):
This was on a system where the RTC was not corrupted and everything was working fine.

````
method return sender=:1.23 -> dest=:1.22 reply_serial=2
array [
dict entry(
string "Timezone"
variant string ""
)
dict entry(
string "LocalRTC"
variant boolean false
)
dict entry(
string "CanNTP"
variant boolean false
)
dict entry(
string "NTP"
variant boolean false
)
dict entry(
string "NTPSynchronized"
variant boolean true
)
dict entry(
string "TimeUSec"
variant uint64 1541004895448222
)
dict entry(
string "RTCTimeUSec"
variant uint64 1541004896000000
)
]
````

I expected to get "true" on "LocalRTC" and "CanNTP".

zagatta-sonah, 2018-11-02:
And I forgot to mention that the RTC (and therefore our vital power-control) keeps working fine when i remove the dtoverlay

agherzan, 2018-11-02:
@zagatta-sonah The documentation you point out sadly is outdated. It is meant for timesyncd which we migrated from and now we are using chronyd as a NTP manager. CC @shaunmulligan 

We are discussing internally a solution to allow a user to activate/deactivate the rtc sync in chronyd (rtcsync) from config.json (or similar). I will update this issue when we have more information.

agherzan, 2018-11-02:
We had a small discussion about this. We currently don't want to disable NTP globally as many components in balenaOS depend on accurate time. We are trying to find a solution to have chrony ignore this specific device.

ZubairLK, 2019-01-22:
@zagatta-sonah chrony writes to the rtc every 11 minutes to keep it in sync.

A possible workaround could be to run `rmmod rtc-pcf8523.ko` from a privileged container. This will remove the rtc module which will prevent chrony from writing to the /dev/rtc0 node.
Of course, that is unwise if you wish to read the time from the rtc sometimes as well. So you'll have to manually read that. And you'll have to manually keep the rtc time in sync.

Also check https://github.com/balena-os/meta-balena/issues/1247#issuecomment-456353808 for a somewhat related thing. Sometimes base images have udev rules that set hwclock. Might want to remove those as well.

ZubairLK, 2019-01-22:
Also made https://github.com/balena-os/meta-balena/pull/1370. Perhaps it is wise to switch to using rtcfile.