balena-os/meta-balena #1279: Flashed Odroid C1/C1+ BalenaOS 2.15.1+rev1.dev image cannot be accessed via SSH

gdeverlant, 2018-11-08:
The board cannot be accessed via SSH although the device seems to have a good heart beat blue LED blinking and recognized by my switch and router.

ZubairLK, 2018-11-08:
@gdeverlant can you share the command you are trying to ssh into the board?
Are you using resin cli `resin ssh` or directly `ssh`?

The default port is 22222 if you are trying `ssh`

gdeverlant, 2018-11-08:
I use BitVise ssh client and port 22

agherzan, 2018-11-08:
As @zubairlk pointed above, you need to use port 22222. See https://github.com/balena-os/meta-balena/pull/1278

gdeverlant, 2018-11-09:
What is the username and password ?

agherzan, 2018-11-09:
For development it's root and empty password.

agherzan, 2018-11-09:
Is this somehow affecting your ssh connection? I'm fail to understand how could that happen.

gdeverlant, 2018-11-09:
Old kernel cannot be used in Docker Swarm mode. The overlay networks are not working. This means Docker is not well supported on this release. I ran the https://raw.githubusercontent.com/moby/moby/master/contrib/check-config.sh

and there is missing a lot of module.

agherzan, 2018-11-09:
I see. I recommend you creating a device specific issue at https://github.com/balena-os/resin-odroid detailing this as much as possible and attaching logs.

Is the ssh working fondle now for you? If yes, I will close this issue and let you follow up with the kernel issues in another thread.

gdeverlant, 2018-11-09:
Yes I was able to connect via ssh.

agherzan, 2018-11-09:
I'm glad. The Kernel issue moved https://github.com/balena-os/resin-odroid/issues/95

closed 2018-11-09 by agherzan