balena-os/meta-balena #1280: resin-image base hash changes

svenschwermer, 2018-11-08:
I occasionally get the following type of error message:
```
ERROR: When reparsing /home/sven/sumo/build/../layers/meta-resin/meta-resin-common/recipes-core/images/resin-image.bb.do_rootfs, the basehash value changed from dbb39e438543b98671b4242e204f1ab5 to db48266ef570a05bfaa65c8a32169f64. The metadata is not deterministic and this needs to be fixed.
ERROR: When reparsing /home/sven/sumo/build/../layers/meta-resin/meta-resin-common/recipes-core/images/resin-image.bb.do_image_complete, the basehash value changed from 25c637125b4ac7134d4539bf25f8b35e to cc8e65bc50c010c499a9a180a983ce41. The metadata is not deterministic and this needs to be fixed.
```
What is this caused by and how can I fix it?

agherzan, 2018-11-08:
I haven't seen this before. Do you have any consistent way to reproduce this?

svenschwermer, 2018-11-08:
> I haven't seen this before. Do you have any consistent way to reproduce this?

Unfortunately, I don't. This is fixed by removing build/tmp and then rebuilding from sstate which doesn't take too long.

agherzan, 2018-11-08:
Yes. That would definitely fix it but I'm definitely curious what changes in resin-image that makes the hashes change as well.

svenschwermer, 2018-11-08:
It could be that I pulled something from upstream and rebuild. Could that explain it?

agherzan, 2018-11-08:
Don't see how that would impact it. At build time things change for the build system - in between parsing and reparsing. Did you change anything while running the build?

svenschwermer, 2018-11-08:
I don't think so. However, I'll close this ticket and I'll reopen once I'm able to reproduce this making sure that nothing was modified while a build is running.

closed 2018-11-08 by svenschwermer

agherzan, 2018-11-08:
Keep us updated. Thanks.

bbinet, 2020-01-30:
I have the exact same issue here. But I don't know (yet) a way to reproduce.