balena-os/meta-balena #1287: Setting host name with Avahi

majorz, 2018-11-14:
Invoking the `SetHostName` method of the `org.freedesktop.Avahi.Server` interface on v2.26 does not succeed (and host name is changed over and over):

```
Nov 14 17:35:27 balena avahi-daemon[734]: Changing host name to 'aaaaaa'.
Nov 14 17:35:27 balena avahi-daemon[734]: Host name conflict, retrying with aaaaaa-2
Nov 14 17:35:28 balena avahi-daemon[734]: Host name conflict, retrying with aaaaaa-3
Nov 14 17:35:29 balena avahi-daemon[734]: Host name conflict, retrying with aaaaaa-4
...
```

Sample code for reproducing this (from a raspberrypi3-python container):
```
import dbus

iface = 'org.freedesktop.Avahi.Server'

host_name = 'aaaaaa'

bus = dbus.SystemBus()

avahi_object = bus.get_object('org.freedesktop.Avahi', '/')
avahi_server_iface = dbus.Interface(avahi_object, iface)

print(avahi_server_iface.GetHostName())

try:
    avahi_server_iface.SetHostName(host_name)
except dbus.exceptions.DBusException as err:
    if err.get_dbus_name() != "org.freedesktop.Avahi.NoChangeError":
        raise err

print(avahi_server_iface.GetHostName())
```

ZubairLK, 2018-11-22:
@majorz @agherzan I think we are hitting an avahi bug here. https://github.com/lathiat/avahi/issues/117

tl;dr. Avahi broadcasts SetHostName on multiple interfaces and basically confuses itself. Thinking the hostname is already assigned and thus conflict.

If I edit `/etc/avahi/avahi-daemon.conf` with `allow-interfaces=enp0s3`, I can set the hostname.

This is a tricky one.. As we probably need Avahi in the container interfaces to work.

Has anyone tried to set the hostname via `org.freedesktop.hostname1` ?

ZubairLK, 2018-11-22:
We can use hostname1 provided by systemd
```
dbus-send --system  --print-reply  --dest=org.freedesktop.hostname1 /org/freedesktop/hostname1 org.freedesktop.hostname1.SetHostname  string:"helloworld2" boolean:true
root@balena:~# hostnamectl 
   Static hostname: test
Transient hostname: helloworld2
         Icon name: computer-vm
           Chassis: vm
        Machine ID: de9f101eeeb2498cb08ecf9d5428f81c
           Boot ID: 0e1b1869ddb54d3db47356ffc2e2f7fa
    Virtualization: oracle
  Operating System: balenaOS 2.24.0+rev3
            Kernel: Linux 4.14.48-yocto-standard
      Architecture: x86-64
root@balena:~# cat /etc/hostname 
test
root@balena:~# hostname
helloworld2
```

agherzan, 2018-11-22:
We expose the hostname via config.json as well. @majorz wouldn't that be good for a workaround?

majorz, 2018-12-04:
There are two workarounds for the Avahi bug until there is a fix from the Avahi maintainers:

1. Do not change the host name with Avahi, but use the `AddAddress` of the `org.freedesktop.Avahi.EntryGroup` instead. It is important to use the no reverse flag when using that, since it will lead to another Avahi bug.

2. If host name really should be changed through Avahi, then first the host name needs to be changed though our supervisor API, then the Avahi daemon should be restarted through the systemd D-Bus API, so that it can pick up the changed host name.

ZubairLK, 2019-01-22:
The thread seems to have reached a conclusion with suitable workarounds.
I don't see anything that can be done via meta-balena.

Closing the issue for cleanup.

mdtusz, 2019-03-28:
@majorz could you elaborate a bit on setting the hostname using `AddAddress` and the no-reverse flag part specifically? There's no formal documentation for usage of avahi with DBus (or at least none I could find) and I can't find any reference to a no-reverse flag either. Setting the hostname using the supervisor seems to have the same issues with a continually incrementing suffix on the hostname/address.

majorz, 2019-08-27:
We should workaround the host name conflict bug somehow as it looks like the original Avahi bug is stale, but we have more and more reports of this.

hedss, 2019-09-05:
AFAIK this was going to be fixed in 0.8 (it was on the roadmap at the time). However, they *were* going to fix it for 0.7... and didn't. I think people had PRd fixes for this...

mgreensmith, 2020-03-06:
This was fixed in avahi `0.8`:

https://github.com/lathiat/avahi/pull/263

Any chance of getting that into the next balenaOS release?

markcorbinuk, 2020-07-06:
I have been unable to reproduce the initially reported behaviour using the following test conditions:

Hardware: RPI3 B+
OS: balenaOS 2.51.1+rev1 (development, Ethernet +Wifi)

Python container built using the following Dockerfile:
```
FROM balenalib/%%BALENA_MACHINE_NAME%%-debian:stretch
RUN apt update && apt install -y dbus python python-dbus
ENV DBUS_SYSTEM_BUS_ADDRESS unix:path=/host/run/dbus/system_bus_socket
CMD ["/bin/bash", "-c", "sleep infinity"]
```

Output of all tests below was monitored by running ```journalctl -f --unit=avahi-daemon``` from a separate OS shell command line.

1) Manually testing the behaviour of setting the Avahi hostname via DBus from the OS command line

The following command works as expected from the OS shell command line:
```dbus-send --system  --print-reply --dest=org.freedesktop.Avahi / org.freedesktop.Avahi.Server.SetHostName  string:"helloworld2" boolean:true```

2) Manually testing the behaviour of setting the Avahi hostname via DBus from inside the Python container

The following command works as expected from inside the container:
```dbus-send --system  --print-reply --dest=org.freedesktop.Avahi / org.freedesktop.Avahi.Server.SetHostName  string:"helloworld2" boolean:true```

3) Testing the behaviour of setting the Avahi hostname via DBus from inside the Python container using the sample Python code supplied in the original post for this issue.

No instability or incrementing of the hostname was observed during any of the above tests.

alexgg, 2020-07-29:
Reproduced on:
```
ID="balena-os"
NAME="balenaOS"
VERSION="2.41.0+rev4"
VERSION_ID="2.41.0+rev4"
PRETTY_NAME="balenaOS 2.41.0+rev4"
MACHINE="raspberrypi4-64"
VARIANT="Development"
VARIANT_ID="dev"
META_BALENA_VERSION="2.41.0"
RESIN_BOARD_REV="94eb7fe"
META_RESIN_REV="747539d"
SLUG="raspberrypi4-64"
```

Running the python code above in a container:
```
Jul 29 11:37:05 4ebb233 avahi-daemon[897]: Host name conflict, retrying with aaaaaa-18
```

However, my initial test on the head of master worked, so I am working my way up.

alexgg, 2020-08-06:
Closing as it is no longer reproducible, re-open if it resurfaces.

commit 7c136fa8 (2020-04-15): avahi: Apply upstream patch not to cache locally generated responses

This fixes a reported problem with hostname clashes when trying to
set the hostname from a container using d-bus.

Fixes #1287

Changelog-entry: Fix setting avahi hostname using dbus from container
Changelog-type: patch
commit 9fd23d32 (2020-04-15): avahi: Apply upstream patch not to cache locally generated responses

This fixes a reported problem with hostname clashes when trying to
set the hostname from a container using d-bus.

Fixes #1287

Changelog-entry: Fix setting avahi hostname using dbus from container
Changelog-type: patch
closed 2020-08-06 by alexgg