balena-os/meta-balena #1289: Unrecognized flag --balenaio for managed build

mschm, 2018-11-15:
The [Custom Build Docs](https://www.balena.io/os/docs/custom-build/#supporting-your-own-board) advise you to specify the `--balenaio` flag in order to create a managed build.

But when I do so with `./resin-yocto-scripts/build/barys -m up-board --balenaio` it fails with an `Unrecognized option --balenaio` error.

What am I missing?

agherzan, 2018-11-15:
Hi @mschm . That is an outdated bit in the documentation. Just completely ignore that flag as it doesn't exist anymore. CC @shaunmulligan

mschm, 2018-11-15:
Thanks @agherzan! 
So how do I select between standalone and managed OS?

agherzan, 2018-11-15:
Right now the images are the same. By default they are unmanaged and you `join` a backend using our CLI. When you download one from the backend, it includes balena-cloud configuration in config.json which makes the os-config to join our backend by default. If you build your own image, the default config.json will not have such a configuration and will default to unmanaged.

agherzan, 2018-11-15:
Created a bug in docs: https://github.com/balena-io/docs/issues/965 . I'll close this issue but feel free to ask any other questions.

closed 2018-11-15 by agherzan

mschm, 2018-11-15:
That's exactly the answer I was hoping for :)

agherzan, 2018-11-15:
I'm glad. Enjoy!