balena-os/meta-balena #1302: `scripts/basic/fixdep: not found` when using kernel headers to compile a kernel module

floion, 2018-11-26:
This was observed on rpi3 and jetson tx2 so far.
Details are here: https://www.flowdock.com/app/rulemotion/resin-devices/threads/LzAzsTCKnbK-R8y07qhncULcdEn

agherzan, 2018-11-26:
Thanks @floion . We will take a loot at this.

blechatellier, 2018-11-26:
Also having that issue on the Intel Nuc

ZubairLK, 2018-11-26:
@blechatellier I'll take a look at this tomorrow morning.

Can you paste the whole output for the build for the nuc image that you see?

Thanks

blechatellier, 2018-11-26:
@ZubairLK thanks, this is the output from `/var/log/nvidia-installer.log` when trying to install NVIDIA drivers with the 4.14.48-yocto-standard kenerl and balenaOS 2.26.0

https://gist.github.com/blechatellier/2d69e6b7c433ecfa79f42aa15e3d0bd8

ZubairLK, 2018-11-27:
https://github.com/balena-io-playground/kernel-module-build/issues/7#issuecomment-442013726

commit f4f898c2 (2018-11-30): kernel-modules-headers: Update to v0.0.12

63baa421 introduced a bug where the tools in the header tarballs were
compiled with the incorrect HOSTCC parameter. As a result the tools
inside the tarball were compiled for x86-64 instead of the native
device architecture. And they had the incorrect ld interpreter path
as well.

Update to v0.0.12 to fix that.
This release also adds objtool support needed by the intel nuc image.

Fixes #1302
Fixes #1303

Change-type: minor
Changelog-entry: Fix kernel module header generation
commit 706829bc (2018-11-30): kernel-modules-headers: Update to v0.0.12

63baa421 introduced a bug where the tools in the header tarballs were
compiled with the incorrect HOSTCC parameter. As a result the tools
inside the tarball were compiled for x86-64 instead of the native
device architecture. And they had the incorrect ld interpreter path
as well.

Update to v0.0.12 to fix that.
This release also adds objtool support needed by the intel nuc image.

Fixes #1302
Fixes #1303

Change-type: minor
Changelog-entry: Fix kernel module header generation
PR #1304 (2018-12-03): Fix kernel header for modules generation
Fixes https://github.com/balena-io-playground/kernel-module-build/issues/7
Fixes #1303 
Fixes #1302 

Tested pi3, jetson and nuc image.
NUC image needs the following workaround 
```
ln -s /lib64/ld-linux-x86-64.so.2  /lib/ld-linux-x86-64.so.2
```
in  build.sh in https://github.com/balena-io-playground/kernel-module-build/

The baselib in balenaOS is `/lib` which is used by the recipe. So the tools built inside the tarball have the interpreter path as  `/lib/ld-linux-x86-64.so.2`. But the baselib in the debian image is `/lib64`. This needs more thought o [... cut]