balena-os/meta-balena #1306: resin-supervisor should depend on dnsmasq

xginn8, 2018-11-28:
`resin-supervisor` fails to update a container when DNS is unavailable, which comes from `dnsmasq`. `resin-supervisor` `Wants=dnsmasq.service`, however I believe it should `Require=dnsmasq.service`. Based on my reading of the `dnsmasq` source, there is no native socket-activated support for `Type=notify` readiness.

agherzan, 2018-11-29:
@xginn8 Wants and Requires adds the dependency in the same way. The only difference is what happens if the dependency fails or can't be added to the transaction. As well, resin-supervisor doesn't depend on dnsmasq. I'm not sure I understand the actual issue here.

ZubairLK, 2020-01-27:
Adding a `Wants=dnsmasq.service` doesn't sound like a bad idea. It should avoid spurious dns errors on first boot for well connected devices.

Don't want to make a hard dependency via Requires.