balena-os/meta-balena #1307: Put files to the /mnt/data/resin-data or /resin-data dirs during build image

ashishka, 2018-11-29:
Hello! 
Can anybody explain me how we can put files to the read/write mounts (/resin-data for example) during build resinOS image with yocto ? Need to put big file, but root fs has size restriction.

Thanks!

agherzan, 2018-11-29:
Hi @ashishka. We don't have this available as a clean option in the build system. It is doable but it's going to be a hack. Could you please detail the use case a little? How is this file used at runtime?

ashishka, 2018-11-29:
Sure, i want make start balenaos with start container more faster, i have plan to do docker image save and put this saved image to the /resinos-data, for example, and make balena load at the first boot, without balena pull, for more faster boot instead balena pull. Is it possible to put this big files during build process ?

agherzan, 2018-11-29:
Is this mecanism you are describing intended for using it with balena cloud services?

ashishka, 2018-11-29:
i was hope trying it without it for first, just have all need images in one balena image, when it flashing to the hdd

agherzan, 2018-11-29:
Right. So for the managed images, the ones you use with balena-cloud, this is already implemented as a balena preload CLI action. See https://www.balena.io/docs/reference/cli/#preload-image- . For the unmanaged case, this is not implemented yet but we are currently working on it and hopefully, we will have it soon.

closed 2018-12-04 by agherzan