balena-os/meta-balena #1337: Investigate htpdate as a fallback if NTP daemons fail

ZubairLK, 2018-12-18:
Seems like there is a utility that can be used to set the time via http. https://linux.die.net/man/8/htpdate
A recipe already exists as well https://layers.openembedded.org/layerindex/recipe/85817/

Investigate it as a fallback.

shaunmulligan, 2019-03-20:
@ZubairLK we've had a few people recently hit some blocked NTP issues and I am wondering if it will be a quick win to prioritise having this fallback

imrehg, 2019-03-21:
The hand-rolled workaround is this line:

```
date -s "$(curl --silent --head --insecure https://api.balena-cloud.com/ping | grep -i Date: | sed 's/[Dd]ate: //')"
```

* using our API endpoint (so no new endpoint needed to be opened on firewalls)
* need to use `--insecure` for people when MitM'd, most of the time it is still correct time, so would be updated anyways even if from the MitM-ig server. And it's using mere header, not data, so security is less of an issue, I think

ZubairLK, 2019-03-21:
Thanks @imrehg . I was mainly thinking of when/how a device can realise its behind a firewall which blocks ntp and then to fallback to htpdate..

imrehg, 2019-03-21:
That's fair too, thanks for the context!

imrehg, 2019-04-29:
bump for internal issue tracker to pick up issue again.

penneydude, 2019-10-08:
Any update on this issue?

We've been having problems with some devices failing to update the system clock due to customer network policies that block NTP.

We've worked around it by using htpdate to keep the system clock in sync in our own containers, but Balena OS updates fail on affected devices due to clock-related SSL validation errors. Our containers with htpdate can't start until the supervisor update has completed, so those devices get stuck in an "updating" state.

ZubairLK, 2019-10-09:
no updates yet I'm afraid.

xginn8, 2020-06-15:
Internal discussion here for using the NITZ time from a GSM modem as a fallback as well: https://www.flowdock.com/app/rulemotion/resin-starters/threads/CishbOHZgrzZ0hkPCBAzc9PqO14

closed 2021-12-01 by bulldozer-balena[bot]