balena-os/meta-balena #1341: Chrony recipe using feature not available in older yocto versions

annymsMthd, 2018-12-19:
The PR #1321 modified the chrony recipe to use bb.utils.filter which is not available is some older versions of yocto, I'm currently using krogoth.

```
ERROR: ExpansionError during parsing /app/build/../layers/meta-balena/meta-resin-krogoth/recipes-core/chrony/chrony_3.2.bb: Failure expanding variable PACKAGECONFIG, expression was editline     ${@bb.utils.filter('DISTRO_FEATURES', 'ipv6', d)}  which triggered exception AttributeError: 'module' object has no attribute 'filter'
```

annymsMthd, 2018-12-19:
https://github.com/balena-os/meta-balena/pull/1321/files#diff-c000e13d95e943c6a43140b2b382cb63R63

ZubairLK, 2018-12-19:
I guess this passed build tests because none of the devices we support balenaCloud on run krogoth anymore.

May I ask which device you are using balenaOS on?

annymsMthd, 2018-12-19:
https://www.compulab.com/products/iot-gateways/iot-gate-imx7-nxp-i-mx-7-internet-of-things-gateway/

I'm currently in talks with them about updating their yocto bsp layer to the newer kernel

ZubairLK, 2018-12-19:
@annymsMthd I'm guessing you might be unable to build network manager as well. That recipe was bumped and also contains `bb.utils.filter`

If none of the devices on balenaCloud run krogoth, we aren't really testing that Yocto version. While this might be a build error, there is no idea about runtime issues. We probably need to deprecate previous Yocto versions.

And in fact, we might not be running `morty` on any device either because that didn't contain bb.utils.filter either

annymsMthd, 2018-12-19:
Understandable. I will look into updating the kernel and yocto version of this board and go from there. Thank you.