balena-os/meta-balena #1344: aufs bug creates unstoppable zombies

pdcastro, 2018-12-22:
* Odroid-C1+ devices with an old kernel (3.10)
* puppeteer leaving orphaned chromium processes
* systemd not properly reaping the zombie because of the old kernel

Suggested solution: backport the following kernel patch from Ben Hutchings, message 47:

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=812207#47

Also, backport the aufs utils patch.

See also:
* https://github.com/moby/moby/issues/18180#issuecomment-193708192
* https://app.frontapp.com/open/cnv_156t7p7

ZubairLK, 2018-12-27:
@spanceac This is for https://github.com/balena-os/balena-odroid

ZubairLK, 2018-12-27:
Moved to odroid as it is a device specific issue. The patches need to be applied there.
https://github.com/balena-os/balena-odroid/issues/101

closed 2018-12-27 by ZubairLK