balena-os/meta-balena #1345: Improvements on default files present in system-connections

majorz, 2018-12-27:
`resin-sample.ignore` contains a `hidden=true` field, which targets hidden networks. We may mark it as false or remove it completely.

The `README.ignore` in the same folder mentions `These files are copied into the filesystem from the boot partition by the resin-net-config script.` We may change `resin-net-config script` to `resin-net-config service` or `service script`.

floion, 2018-12-27:
Why remove the option to connect to hidden SSIDs? People have asked for this a while back, that is why this option is present there.

majorz, 2018-12-27:
I reported this as a user was trying to use the file, but he could not connect to a regular network. If we set `hidden=false` users who are looking to connect to a hidden network will know what to change. Common users may not be aware of what hidden means, leave the field set to true and not be able to connect.

floion, 2018-12-27:
Having hidden=true does not prevent connecting to non-hidden networks afaik.