balena-os/meta-balena #1348: Private DNS bypassed in favour of 8.8.8.8

chrisallmark, 2019-01-02:
Since upgrading to BalenaOS v2.27.0 I have a widespread failure across my entire fleet as my private DNS gets intermittently bypassed and 8.8.8.8 is used causing name resolution failures. I believe this is directly related to https://github.com/balena-os/meta-balena/issues/1293

agherzan, 2019-01-02:
@charleswhchan what is the error you are getting when trying to resolve a local domain name? dnsmasq should retry an available server when 8.8.8.8 can't resolve. Do you have any device that shows this failure consistently?

chrisallmark, 2019-01-02:
@agherzan I use the following configuration on my PI's:

[ipv4]
dns=10.184.10.30;10.184.10.20;10.184.10.10;
ignore-auto-dns=true
method=auto

I don't ever want to use 8.8.8.8 as it won't resolve our private DNS entries. It happens intermittently but frequently (i.e. multiple time per day) across every device which is referencing one of our private DNS entries. I raised this with support before Christmas and got the following (possible) explanation:

"I can only theorize about what happens. The DNS requests are sent to all servers - not only to one of those. So it is either that the none of the DNS servers were reachable at the moment of failure. Or there is a racing condition with our DNS client - e.g. a response from Google comes first and is taken as the valid one." - Zahari from Balena.io

I'm raising this a formal issue as I need to get an ETA on fix.

agherzan, 2019-01-02:
@charleswhchan thanks for the summary. I had a short chat with @majorz as well and understood the issue. We will try to reproduce it and update you as soon as we have some information. At first glance, it looks like an abnormal dnsmasq behaviour.

agherzan, 2019-01-02:
A small update on this issue. We investigated it and after reproducing the issue we figured out why this happens and we realized that it actually works accordingly to dnsmasq's design. And I'll explain.
The way this service works is using as a list of "known" servers. This list is queried by the daemon when it receives the first request at which point it fires a DNS request to each of these known servers. The first server which replies is used for the next queries for a while (based on an internal algorithm). The problem is that it considers negative replies as well so in the case where 8.8.8.8 replies first (with a negative reply for a local domain name) dnsmasq will save 8.8.8.8 as the "active" server and will return its reply for a while (that has a twist though because it actually caches the reply by default - see `-N` in http://www.thekelleys.org.uk/dnsmasq/docs/dnsmasq-man.html). When this active server cache expires, there will be another chance where the servers will race for a reply. You can read more on this behaviour/design/server algorithm here: http://lists.thekelleys.org.uk/pipermail/dnsmasq-discuss/2009q3/003295.html.

It seems that this is a known issue upstream (see for example https://bugs.launchpad.net/ubuntu/+source/network-manager/+bug/1003842 for example) and it is considered to be a bad network design because it breaks the assumption made by dnsmasq that all the servers it knows are identical in terms of their replies. That is indeed the case here because the local server has specific entries that the upstream DNS servers don't know about.

Hope this gives you a context. We will discuss internally on how to properly fix this and will keep you updated.

samuelcasas, 2019-01-04:
Any update yet? I guess a temporary workaround will be to either edit `/etc/dnsmasq.d/balena.conf` or compile a custom image without DNS fallback, but compile a custom image would be overkill so I'll try to stick to the first option

agherzan, 2019-01-08:
We will fix this by taking advantage of https://github.com/balena-os/meta-balena/blob/master/README.md#dnsservers in the following way:

- first of all, we will remove the hardcoded fall back in the dnsmasq configuration
- when dnsServers is not defined, or empty, the default value of it will be considered google DNS (8.8.8.8)
- when dnsServers is "null", the system will not alter dnsServers to the default server as stated above
- when dnsServers is defined and not "null" it will be considered as such without any fallback servers

@samuelcasas @chrisallmark What do you guys think?

closed 2019-01-28 by agherzan

epers, 2019-05-21:
Hi, is there a way to configure this without rebuilding the entire image? I tried putting it in config.json but I'm still seeing it hit 8.8.8.8 on 2.32.0+rev1.
If there's a way to set this via a configuration envvar that'd be even better.

agherzan, 2019-05-22:
The current way to configure this is using the config.json's object as defined here https://github.com/balena-os/meta-balena/blob/master/README.md#dnsservers. This allows you to avoid 8.8.8.8 default DNS server. There is no support for environment variables at this point for this configuration.

epers, 2019-05-22:
Okay, then I must be doing something wrong, can you take a quick look at my config? Based on the documentation and our config below I'd expect it to use 10.10.10.10 exclusively, but it's also still using quad8 and the dns servers it receives from dhcp. Putting `"null"`, `"'null'"`, or `'"null"'` in there didn't stop it from using quad8 either. 

I can take this to the forums or open a new issue if I'm making too much noise on an old thread. Just figured I'd get right to the people who knew about this particular bit.

```
{
    "apiEndpoint": "https://api.balena.our_fqdn",
    "appUpdatePollInterval": 600000,
    "applicationId": 2,
    "applicationName": "mpiDisplay-v1-dev",
    "balenaRootCA": "<<snip>>",
    "deltaEndpoint": "https://delta.balena.our_fqdn",
    "deviceApiKey": "<<snip>>",
    "deviceApiKeys": {
        "api.balena.our_fqdn": "<<snip>>"
    },
    "deviceType": "raspberrypi3",
    "listenPort": 48484,
    "mixpanelToken": "__unused__",
    "pubnubPublishKey": "",
    "pubnubSubscribeKey": "",
    "registryEndpoint": "registry.balena.our_fqdn",
    "userId": 2,
    "username": "root",
    "ntpServers": "time.balena.our_fqdn",
    "dnsServers": "10.10.10.10",
    "os": {
        "sshKeys": ["ssh-rsa <<snip>> balena@balena.our_fqdn"]
    },
    "vpnEndpoint": "vpn.balena.our_fqdn",
    "vpnPort": 443,
    "wifiKey": "<<snip>>",
    "wifiSsid": "balena",
    "uuid": "40a5e2bb6f92da1446d6a4edc674c759",
    "registered_at": 1558474220872,
    "deviceId": 6
}
```
```
root@40a5e2b:~# systemctl status dnsmasq
● dnsmasq.service - DNS forwarder and DHCP server
   Loaded: loaded (/lib/systemd/system/dnsmasq.service; enabled; vendor preset: enabled)
  Drop-In: /etc/systemd/system/dnsmasq.service.d
           └─dnsmasq.conf
   Active: active (running) since Wed 2019-05-22 11:18:24 UTC; 33min ago
  Process: 790 ExecStartPre=/usr/bin/dnsmasq --test (code=exited, status=0/SUCCESS)
 Main PID: 794 (dnsmasq)
   CGroup: /system.slice/dnsmasq.service
           └─794 /usr/bin/dnsmasq -x /run/dnsmasq.pid -a 127.0.0.2,10.114.102.1 -7 /etc/dnsmasq.d/ -r /etc/resolv.dnsm>

May 22 11:18:24 40a5e2b dnsmasq[794]: dnsmasq[794]: DBus support enabled: connected to system bus
May 22 11:18:24 40a5e2b dnsmasq[794]: dnsmasq[794]: reading /etc/resolv.dnsmasq
May 22 11:18:24 40a5e2b dnsmasq[794]: dnsmasq[794]: using nameserver 8.8.8.8#53
May 22 11:18:24 40a5e2b dnsmasq[794]: dnsmasq[794]: read /etc/hosts - 6 addresses
May 22 11:18:24 40a5e2b dnsmasq[794]: dnsmasq[794]: using nameserver 10.10.10.10#53
May 22 11:18:24 40a5e2b dnsmasq[794]: dnsmasq[794]: using nameserver 8.8.8.8#53
May 22 11:18:33 40a5e2b dnsmasq[794]: dnsmasq[794]: reading /etc/resolv.dnsmasq
May 22 11:18:33 40a5e2b dnsmasq[794]: dnsmasq[794]: using nameserver 10.10.10.10#53
May 22 11:18:33 40a5e2b dnsmasq[794]: dnsmasq[794]: using nameserver 10.15.40.21#53
May 22 11:18:33 40a5e2b dnsmasq[794]: dnsmasq[794]: using nameserver 10.15.40.22#53
```

agherzan, 2019-06-02:
@epers Right. I realised now what the issue is. The change that introduced the mechanism I pointed to above was included from 2.30.0 but you are running a version before that. Can you try to reproduce on a 2.30.0 version or a later one?

SteeveGL, 2020-11-18:
This issue still exist in 2.60.1

I noticed the warning message. I did not modify that file.
```
root@66da9dd:~# systemctl status dnsmasq
Warning: The unit file, source configuration file or drop-ins of dnsmasq.service changed on disk. Run 'systemctl>
● dnsmasq.service - DNS forwarder and DHCP server
   Loaded: loaded (/lib/systemd/system/dnsmasq.service; enabled; vendor preset: enabled)
  Drop-In: /etc/systemd/system/dnsmasq.service.d
           └─dnsmasq.conf
   Active: active (running) since Tue 2020-11-17 23:57:15 UTC; 43s ago
  Process: 1408 ExecStartPre=/usr/bin/dnsmasq --test (code=exited, status=0/SUCCESS)
 Main PID: 1414 (dnsmasq)
    Tasks: 1 (limit: 2300)
   Memory: 728.0K
   CGroup: /system.slice/dnsmasq.service
           └─1414 /usr/bin/dnsmasq -x /run/dnsmasq.pid -a 127.0.0.2,10.114.102.1 -7 /etc/dnsmasq.d/ -r /etc/reso>

Nov 17 23:57:19 66da9dd dnsmasq[1414]: dnsmasq[1414]: reading /etc/resolv.dnsmasq
Nov 17 23:57:19 66da9dd dnsmasq[1414]: dnsmasq[1414]: using nameserver 8.8.8.8#53
Nov 17 23:57:19 66da9dd dnsmasq[1414]: dnsmasq[1414]: using nameserver 10.0.0.10#53
Nov 17 23:57:19 66da9dd dnsmasq[1414]: dnsmasq[1414]: using nameserver fd60:7016:bac7::1#53
Nov 17 23:57:38 66da9dd dnsmasq[1414]: dnsmasq[1414]: reading /etc/resolv.dnsmasq
Nov 17 23:57:38 66da9dd dnsmasq[1414]: dnsmasq[1414]: using nameserver 8.8.8.8#53
Nov 17 23:57:38 66da9dd dnsmasq[1414]: dnsmasq[1414]: using nameserver 10.0.0.10#53
Nov 17 23:57:38 66da9dd dnsmasq[1414]: dnsmasq[1414]: using nameserver fd60:7016:bac7::1#53
Nov 17 23:57:38 66da9dd dnsmasq[1414]: dnsmasq[1414]: using nameserver 74.116.184.28#53
Nov 17 23:57:38 66da9dd dnsmasq[1414]: dnsmasq[1414]: using nameserver 74.116.184.29#53
```