balena-os/meta-balena #1366: Trim kernel modules

ZubairLK, 2019-01-16:
There might be some that we don't even enable the drivers for but do ship in the OS.

Audit all the modules we include in the OS and trim as many as possible.

pdcastro, 2019-01-18:
To confirm, the need for this work comes from the fact that there's not enough disk space to include the required WiFi firmware for some built-in NUC chipsets.

Forum thread reporting the problem on balenaOS 2.29.0+rev1:
https://forums.balena.io/t/intel-nuc-wifi-doesnt-work/5041/6

rgaufman, 2019-02-03:
Any update on this? :)

floion, 2019-02-04:
Hi. We have included the wifi firmware in the Intel NUC image, if that is what you are interested in.

rgaufman, 2019-02-04:
Ah, no I was following up on adding coretemp and nct6775 for temperature monitoring on NUC and similar, the issue is here but it references this one as a solution: https://github.com/balena-os/balena-intel/issues/163

ZubairLK, 2019-03-11:
https://github.com/balena-os/meta-balena/pull/1418 addressed this and we've freed up ~30-40 megs in the rootfs.

closed 2019-03-11 by ZubairLK

ZubairLK, 2019-03-11:
I notice a large number of `CONFIG_DVB_*` enabled in the kernel and being shipped on the nuc image..

Not entirely sure how useful they are or if they are used by anyone.

ZubairLK, 2019-03-11:
Also, before I forget. https://github.com/balena-os/balena-intel/pull/161 can free up 8.6 megs on many devices. Only did that for nuc at the time as we were reaching the limit for the nuc.