balena-os/meta-balena #1367: Persistent logs seem to disappear and reappear

spanceac, 2019-01-17:
Enable persistent logging on a board.

Check the output of `journalctl --list-boots` and then reboot.

After one ore more reboots you will see less boot logs appear that reboots.

Sometimes they even reappear.

ZubairLK, 2019-01-21:
I took a look at this.

Its a Heisen-bug so hard to tell exactly what is going on.

Did find one subtle thing to fix https://github.com/balena-os/meta-balena/pull/1369

It is also possible the forceful reboots by the supervisor prevented the journal to flush to the disk: https://github.com/balena-io/balena-supervisor/issues/861 That issue has been fixed in the supervisor now.

But I think people are hitting the 8 megabyte limit. And the journal is rotating over itself as well. 8M is just enough to hold a few reboots. If there is a chatty service/application, 8M will disappear quickly.
I tested on the tx2. 8M is just enough for 3-4 reboots. With 'no' application.

ZubairLK, 2019-01-23:
I don't think there is much here except what has been said above.
I'd say we can close this issue.

agherzan, 2019-01-23:
@ZubairLK I suspect that there is something else here as well. Have you tried without a user application? We shouldn't have that many logs in the journal and a fair amount of boot logs should be easily handled. I think I've seen this issue in the past and it was somehow connected to machine id - the logs are stored per machine id and I was ending up with multiple ones.

ZubairLK, 2019-01-24:
The scenario where this happens is if the initramfs fails to mount the state partition. And fails to bind mount /etc/machine-id from the state partition. Then systemd generates a new random machine-id that doesn't persist if the next time the initramfs succeeds in mounting the state partition.

There is a 30 second timeout here.
https://github.com/balena-os/meta-balena/blob/609118963ab642a858469202d36538a7978fa1b2/meta-resin-common/recipes-core/initrdscripts/files/machineid#L13