balena-os/meta-balena #1387: Add ability to disable serial console on "development" variants

shaunmulligan, 2019-02-08:
Often times people are trying to use the serial console with sensors, etc and having the console automatically attach to it can cause all kinds of issues and make things difficult to get working.

agherzan, 2019-02-08:
@shaunmulligan would this be a build time feature or a runtime one?

ZubairLK, 2019-02-08:
@agherzan runtime from what I understood

shaunmulligan, 2019-02-13:
@agherzan and @ZubairLK I would be happy having it as a boot time thing, like set in the config.json or something.

ajlennon, 2021-04-25:
Did this every happen? It would help me out too as I have exactly this problem

nmaas87, 2021-07-03:
Just run into this issue, while I was using the development version - the daemon seems to have stolen my serial interface (which I used for a GPS mouse) again and again. Would be nice to have a switch to disable this behaviour, especially when developing connected hardware locally :)

nmaas87, 2021-07-03:
I used this ugly hack to disable the serial console, it is really ugly, but it somewhat works for me...
So in a pinch ;)...

````
mount -o remount,rw /
echo 'dwc_otg.lpm_enable=0 rootwait cgroup_enable=memory' > /mnt/boot/cmdline.txt
systemctl stop serial-getty@serial0.service
systemctl disable serial-getty@serial0.service
systemctl stop serial-getty@ttyS0.service
systemctl disable serial-getty@ttyS0.service
systemctl mask serial-getty@serial0.service
systemctl stop serial-getty@serial0.service
systemctl mask getty@tty1.service
systemctl stop getty@tty1.service
reboot
````

alexgg, 2021-09-30:
This has been addressed in https://github.com/balena-os/meta-balena/issues/2226 available from v2.85.0 onwards.

ajlennon, 2021-09-30:
Oh great! Thanks for that

alexgg, 2021-09-30:
Actually, the feature in v2.85.0 is that there is no more development variant but a runtime switch for development mode. But still a single switch, no independent serial console control. I am re-opening this, sorry for the noise.

haata, 2022-11-11:
For those running into this silly issue (for me it's with a cell modem attached to ttyS0...).
This can help https://github.com/edorgeville/balena-dbus-disable-getty-serial

alexgg, 2022-11-11:
thanks for sharing @haata

julianoes, 2024-10-03:
@haata  This can help https://github.com/edorgeville/balena-dbus-disable-getty-serial

For some reason this does not always seem to work for me, and sometimes the console still ends up running.

This is really frustrating @alexgg! I really believe that a flag called `enable_uart` should not define whether a console is running or not, that should be orthogonal to the UART being active. In our example that's the connection to a microcontroller and the console running on it screws up everything.

alexgg, 2024-10-03:
hey @julianoes may I ask you to raise this through our [support channels](https://www.balena.io/support) so we can help? BalenaOS supports multiple device types, typically with multiple UARTS, so the console can have a dedicated one while other peripherals like microcontrollers can use another one. Also, the `enable_uart` you refer to is probably specific to the RaspberryPi's `config.txt` and it configures the on-board firmware to enable the UART so there is not much that can be done from the OS there.

In any case, if you reach out via support and provide us with enough details about your design I am confident we can provide a working solution.

julianoes, 2024-10-03:
@alexgg I will do that, thanks.

That being said, I still think balena's way of inferring a console from a `enable_uart` setting is fundamentally wrong. One is a board device, the other one is a debug option. The fact that `enable_uart` causes a console to be started, is not intuitive and - I bet - causes a lot of confusion for users coming from Raspberry Pi OS.

julianoes, 2024-10-03:
@alexgg I'm not sure how to raise this in the forums, there are already several threads about it, so starting a new one probably won't help:
https://forums.balena.io/t/disable-console-over-serial-in-dev-on-rpi3/1412/44
https://forums.balena.io/t/raspberry-pi-4-uart-link-fails-after-approx-5s/306933/9
https://forums.balena.io/t/serial-interface-development-image/6099/6

It's clearly an issue re-appearing again and again.