balena-os/meta-balena #1391: Healthchecking/watchdoging containerd

imrehg, 2019-02-12:
There can be cases when `balena-engine-containerd` stops working, but `balena-engine` does not, and healthchecks don't catch this currently. Seems like in these situations stopping then restarting the `balena` service restores the functionality, thus likely a watchdog/healthcheck would allow recovery. Should have some checks for these situations.

[Forums](https://forums.balena.io/t/what-does-status-unhealthy-mean-and-why-wont-this-container-pause/5279)

robertgzr, 2019-03-12:
@ZubairLK we just merged https://github.com/balena-os/balena-containerd/commit/573d758a006c46467c8c16d4025b6d5b55bebfc1 that could be used to check containerd's health and reload balenaEngine (solving https://github.com/balena-os/balena-engine/issues/142)

```shell
balena-engine-containerd-ctr \
	--address /var/run/balena-engine/containerd/balena-engine-containerd.sock \
    --namespace moby
	--connect-timeout n \ # seconds
	--timeout n \ # seconds
	health
[ $? -eq 0 ] && echo "containerd healthy!"
```

ZubairLK, 2019-03-12:
@robertgzr great work!

Which balena-engine version will this get merged into?

robertgzr, 2019-03-12:
thanks for the reminder, I will backport it to the `17.12` branch

ZubairLK, 2019-05-21:
bump. saw a hung container on support on 17.12

commit 69ab09cb (2019-07-11): balena: Add another healthcheck

Occasionally balena ps and balena info work but the balena daemon is
unable to start a new container. This is usually when something in runc
or containerd is not functioning correctly.

Add a healthcheck to spin up a simple hello-world container as well.

The trade-off here is that there will be a few extra writes to disk
every healthcheck timeout. But there will be a benefit that if runc
or containerd is in a bad state, the healthcheck will fail and systemd
will restart the balena daemon to recover the application container

Fixes #1391

Change-type: patch
Cha [... cut]
commit 2e447843 (2019-07-11): balena: Add another healthcheck

Occasionally balena ps and balena info work but the balena daemon is
unable to start a new container. This is usually when something in runc
or containerd is not functioning correctly.

Add a healthcheck to spin up a simple hello-world container as well.

The trade-off here is that there will be a few extra writes to disk
every healthcheck timeout. But there will be a benefit that if runc
or containerd is in a bad state, the healthcheck will fail and systemd
will restart the balena daemon to recover the application container

Fixes #1391

Change-type: patch
Cha [... cut]
PR #1392 (2019-07-15): balena: Add another healthcheck
Occasionally balena ps and balena info work but the balena daemon is
unable to start a new container. This is usually when something in runc
or containerd is not functioning correctly.

Add a healthcheck to spin up a simple hello-world container as well.

The trade-off here is that there will be a few extra writes to disk
every healthcheck timeout. But there will be a benefit that if runc
or containerd is in a bad state, the healthcheck will fail and systemd
will restart the balena daemon to recover the application container

Fixes #1391

Change-type: patch
Chan [... cut]
commit 15169084 (2019-07-31): balena: Add another healthcheck

Occasionally balena ps and balena info work but the balena daemon is
unable to start a new container. This is usually when something in runc
or containerd is not functioning correctly.

Add a healthcheck to spin up a simple hello-world container as well.

The trade-off here is that there will be a few extra writes to disk
every healthcheck timeout. But there will be a benefit that if runc
or containerd is in a bad state, the healthcheck will fail and systemd
will restart the balena daemon to recover the application container

Fixes #1391

Change-type: patch
Cha [... cut]
closed 2019-07-31 by agherzan