balena-os/meta-balena #1395: Blacklisted WiFi/BT drivers for balenaFin are still loaded

majorz, 2019-02-15:
We disable the in-kernel WiFi/BT Marvell drivers in favor of the proprietary ones, but those are still loaded although they are blacklisted.

```
root@balena:~# cat /etc/modprobe.d/blacklist.conf
blacklist mwifiex
blacklist mwifiex_sdio
blacklist btmrvl
blacklist btmrvl_sdio
```

```
root@balena:~# lsmod
Module                  Size  Used by
bnep                   20480  2
ipt_REJECT             16384  1
nf_reject_ipv4         16384  1 ipt_REJECT
mwifiex_sdio           36864  0
mwifiex               294912  1 mwifiex_sdio
btmrvl_sdio            28672  0
btmrvl                 20480  1 btmrvl_sdio
ipt_MASQUERADE         16384  3
nf_nat_masquerade_ipv4    16384  1 ipt_MASQUERADE
br_netfilter           24576  0
nf_conntrack_netlink    40960  0
nfnetlink              16384  3 nf_conntrack_netlink
xt_owner               16384  0
i2c_dev                16384  0
sch_fq_codel           20480  11
smsc95xx               32768  0
usbnet                 36864  1 smsc95xx
ftdi_sio               40960  0
snd_bcm2835            32768  0
mii                    16384  2 usbnet,smsc95xx
usbserial              40960  1 ftdi_sio
spidev                 16384  0
rtc_ds1307             24576  0
hwmon                  16384  1 rtc_ds1307
snd_pcm                98304  1 snd_bcm2835
snd_timer              32768  1 snd_pcm
snd                    69632  3 snd_timer,snd_bcm2835,snd_pcm
i2c_bcm2835            16384  0
spi_bcm2835            16384  0
uio_pdrv_genirq        16384  0
i2c_gpio               16384  0
uio                    20480  1 uio_pdrv_genirq
i2c_algo_bit           16384  1 i2c_gpio
fixed                  16384  0
sd8xxx                581632  2
mlan                  450560  1 sd8xxx
cfg80211              589824  2 mwifiex,sd8xxx
bt8xxx                 86016  1
bluetooth             376832  27 btmrvl_sdio,bt8xxx,bnep,btmrvl
ecdh_generic           28672  1 bluetooth
rfkill                 32768  6 bluetooth,cfg80211
```

ZubairLK, 2019-02-18:
@majorz I am unable to reproduce this on v2.29.2 dev image.

```
root@balena:~# cat /etc/os-release
ID="balena-os"
NAME="balenaOS"
VERSION="2.29.2+rev2"
VERSION_ID="2.29.2"
PRETTY_NAME="balenaOS 2.29.2+rev2"
RESIN_BOARD_REV="c561919"
META_RESIN_REV="22fffcd"
SLUG="fincm3"
MACHINE="fincm3"
VARIANT="Development"
VARIANT_ID="dev"
root@balena:~# cat /etc/modprobe.d/blacklist.conf
blacklist mwifiex
blacklist mwifiex_sdio
blacklist btmrvl
blacklist btmrvl_sdio
root@balena:~# lsmod
Module                  Size  Used by
bnep                   20480  2
ipt_REJECT             16384  1
nf_reject_ipv4         16384  1 ipt_REJECT
ipt_MASQUERADE         16384  3
nf_nat_masquerade_ipv4    16384  1 ipt_MASQUERADE
br_netfilter           24576  0
nf_conntrack_netlink    40960  0
nfnetlink              16384  3 nf_conntrack_netlink
xt_owner               16384  0
i2c_dev                16384  0
smsc95xx               32768  0
ftdi_sio               40960  0
usbnet                 36864  1 smsc95xx
mii                    16384  2 usbnet,smsc95xx
usbserial              40960  1 ftdi_sio
rtc_ds1307             24576  0
hwmon                  16384  1 rtc_ds1307
i2c_gpio               16384  0
uio_pdrv_genirq        16384  0
i2c_algo_bit           16384  1 i2c_gpio
uio                    20480  1 uio_pdrv_genirq
fixed                  16384  0
sch_fq_codel           20480  7
sd8xxx                475136  1
mlan                  442368  1 sd8xxx
cfg80211              589824  1 sd8xxx
bt8xxx                106496  1
bluetooth             376832  25 bt8xxx,bnep
ecdh_generic           28672  1 bluetooth
rfkill                 32768  5 bluetooth,cfg80211
root@balena:~# lsmod | grep mwifi
root@balena:~# lsmod | grep btmr
root@balena:~# nmcli c s
NAME                UUID                                  TYPE      DEVICE      
PandoraBox_7C67C6   c41a453d-3319-4b5f-87af-97ba7c7df16b  wifi      wlan0       
Wired connection 1  e137804d-4463-3f5b-b3cc-31b306d5eae8  ethernet  eth0        
supervisor0         58140a3f-7982-47e1-a8c1-7da57141a143  bridge    supervisor0 
root@balena:~# ifconfig wlan0
wlan0     Link encap:Ethernet  HWaddr AC:3F:A4:DA:DB:EA  
          inet addr:192.168.1.107  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::e470:1df7:3cca:90b2/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:292 errors:0 dropped:0 overruns:0 frame:0
          TX packets:116 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:51766 (50.5 KiB)  TX bytes:19095 (18.6 KiB)

root@balena:~# 
```

closed 2019-02-18 by agherzan

majorz, 2019-02-18:
Marking it as invalid as I could not reproduce this now as well on two different OS versions, and one is prod, the other - dev.