balena-os/meta-balena #1396: Investigate cpu performance

ZubairLK, 2019-02-18:
It would be nice to investigate cpu/io/mem/network performance and then compare it to standard distributions on devices. There might be overheads we have that we can reduce.

ZubairLK, 2019-02-18:
pi0w, no app container, 2.29.2

even after 5 minutes, the supervisor and balena daemon seem to be doing something

```
Mem: 254004K used, 237460K free, 3864K shrd, 19484K buff, 100412K cached
CPU:  87% usr  12% sys   0% nic   0% idle   0% io   0% irq   0% sirq
Load average: 2.07 2.24 1.62 2/143 1510
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
 1081  1069 root     R     154m  32%  62% node /usr/src/app/dist/app.js
  782     1 root     S     873m 182%  25% /usr/bin/balenad --experimental --log-driver=journald -s aufs -H fd:// -H unix:///var/run/balena.sock -H unix:///var/run/balena-engine.sock -H
  647     1 root     S     9868   2%   5% @sbin/plymouthd --tty=tty1 --mode=boot --pid-file=/run/plymouth/pid --attach-to-session --kernel-command-line=plymouth.ignore-serial-consoles s
 1487   666 root     R     2992   1%   4% top
  780     1 root     S     857m 178%   1% /usr/bin/balenad --delta-data-root=/mnt/sysroot/active/balena --delta-storage-driver=aufs --log-driver=journald -s aufs --data-root=/mnt/sysroo
  848   782 root     S     856m 178%   1% balena-engine-containerd --config /var/run/balena-engine/containerd/containerd.toml
```

ZubairLK, 2019-02-19:
So cpu load is unusually high on my pi0 on balenaOS
`15:09:46  up   0:15,  1 user,  load average: 0.78, 0.84, 0.66`

Its sensible on pi0 on raspbian lite. 
`15:16:54 up 21:29,  2 users,  load average: 0.07, 0.19, 0.16 `
I have a docker container running a sleep on both.

I'm going to try and bring down various services in the os to see what is causing it.

ZubairLK, 2019-02-22:
note to self: check out health monitoring using netdata.

Tom-Julux, 2022-08-08:
I investigated a little bit and the high initial cpu usage (at least on my system - RPI3) originates from rngd.

Further investigation yielded that for some reason the jitter source was used, even when the hardware randomness source hwrng is available. To solve this issue one could modify the command to run rngd from `/usr/sbin/rngd -f -r /dev/hwrng` from `/usr/sbin/rngd -f -r /dev/hwrng -x jitter` on raspberry pi systems.

Sadly I know exactly nothing about yocto, so I cannot implement this myself

alexgg, 2022-08-31:
hi @Tom-Julux what I think happens is that all entropy sources need to be initialized before the best performing (hardware) engine is chosen. So the jitter source even if it won't be used also has to be initialized (unless you pass the `-x jitter`) option, even if it won't be used as there is a hardware entropy source available. I will look into adding the `-x entropy` to device types that have a hardware entropy source to increase boot time.

srd424, 2024-03-11:
It might be worth pulling a new version of rngd, too - my pi zeros still have 6.15 despite being on the latest production BalenaOS, 6.16 changelog includes "Fix jitterentropy long timeout failures on low power hardware,"

alexgg, 2024-03-11:
hey @srd424 the version of `rngd` comes from Poky Kirkstone - we will update as part of updating to the new Scarthgap LTS release which is scheduled to release in April.