balena-os/meta-balena #1398: Custom certificates handled system-wide

agherzan, 2019-02-19:
For various features (for example balena on-prem) we support having a custom CA in the `config.json` configuration file. At this point, we handle this certificate at the level of the clients (it's handled correctly for example by `os-config`) but not available system-wide through OpenSSL interface. We want to start handling this system-wise (so we don't have to configure each and every client we use) as it follows:

- the OS will try to regenerate the certificates based on `config.json` and host's certificates
- this generation when changes detected, will regenerate the hierarchy of hash links and the bundle file

imrehg, 2019-05-21:
Ping so gets into internal issue tracker.

imrehg, 2019-05-21:
@agherzan another use case, for the `redsocks` proxy to use a custom CA too, as in the linked support thread above

imrehg, 2019-07-17:
Another use `curl` needs this across the OS, for example for `resin-device-progress` as well

imrehg, 2019-07-17:
Also mentioning from this [Flowdock thread](https://www.flowdock.com/app/rulemotion/r-resinos/threads/7TM1YVOv-l5DzvE_TcwxoBNr0IU), that that entry might contain multiple certificates, and thus might be relevant for other components like os-config:

A CA file can have multiple certificates, so BALENA_ROOT_CA can actually contain multiple certificates. The The `NODE_EXTRA_CA_CERTS`  docs at https://nodejs.org/api/cli.html#cli_node_extra_ca_certs_file says

> The file should consist of one or more trusted certificates in PEM format

So that already handles the part, while https://github.com/balena-os/os-config might need improvements too.

There were some ideas to parse the settings from `config.json` and alias curl for curl calls to work better overall, as most of the cases when it's needed, ie. the API communications are through curl in the OS. That's just some contextual note.

commit 28d68725 (2020-07-27): Respect balenaRootCA system-wide

We allow the user to specify a custom CA in the .balenaRootCA key
of config.json but at this moment this is only respected by
the supervisor. This commit adds it to the system-wide CA bundle
so that the CA is respected everywhere.

Fixes #1398

Change-type: minor
commit 3769842a (2020-08-20): Respect balenaRootCA system-wide

We allow the user to specify a custom CA in the .balenaRootCA key
of config.json but at this moment each tool has to implement support
if it wants to use it. This commit adds it to the system-wide CA bundle
so that the CA is respected everywhere.

Fixes #1398

Change-type: minor
commit d68d4208 (2020-08-20): Respect balenaRootCA system-wide

We allow the user to specify a custom CA in the .balenaRootCA key
of config.json but at this moment each tool has to implement support
if it wants to use it. This commit adds it to the system-wide CA bundle
so that the CA is respected everywhere.

Fixes #1398

Change-type: minor
commit 02bca97e (2020-08-20): Respect balenaRootCA system-wide

We allow the user to specify a custom CA in the .balenaRootCA key
of config.json but at this moment each tool has to implement support
if it wants to use it. This commit adds it to the system-wide CA bundle
so that the CA is respected everywhere.

Fixes #1398

Change-type: minor
closed 2020-09-05 by bulldozer-balena[bot]