balena-os/meta-balena #1403: Investigate plymouth

ZubairLK, 2019-02-22:
While investigating #1390  and #1396 , we found that plymouthd via `plymouth-start.service` is always running, eats quite a bit of memory(~10mb) and cpu on weaker devices such as the pi0.

Investigate
- a different splash daemon
- any compile time config options that can reduce its usage
- any way we can kill plymouthd and not have always running. Perhaps configurable in case it is still needed to keep the logo up all the time etc.

ZubairLK, 2019-03-08:
Couldn't find an easy alternative. psplash exists. But it compiles the splash logo in the binary itself. We can't really use that easily.

I do notice one thing. raspbian lite runs `plymouth-quit.service` to quit the daemon. almost the last systemd service before `multi-user.target`. That kills the daemon. We do ship that service. It is marked enabled but remains dead/inactive.
```
root@62d0b9a:~# systemctl status plymouth-quit
● plymouth-quit.service - Terminate Plymouth Boot Screen
   Loaded: loaded (/lib/systemd/system/plymouth-quit.service; static; vendor preset: enabled)
   Active: inactive (dead)
root@62d0b9a:~#
```

which leads to this. We'd like the boot splash to remain until the application takes over..
https://github.com/balena-os/meta-balena/commit/e32029ceafc274470628d6bd767844f797ca0653

a) have some mechanism for headless boards to enable the `plymouth-quit.service` to reduce memory/resource. (or disable plymouth entirely. its headless..
b) have some mechanism so that when an application takes over the framebuffer, plymouth is killed off.

ZubairLK, 2019-03-25:
Can't seem to find any lightweight alternative to plymouth that reads the logo at runtime.
`psplash` is lightweight but the png is built into the binary during the build.

floion, 2020-05-28:
Related: https://jel.ly.fish/40e4e59d-a29e-4ba2-8fd2-3c74715af0d1

alexgg, 2020-06-03:
Reference reads:
* https://blogs.gnome.org/halfline/2009/11/28/plymouth-%E2%9F%B6-x-transition/
* https://gitlab.freedesktop.org/plymouth/plymouth/-/merge_requests/28