balena-os/meta-balena #1423: Enabling persistent logging sometimes eventually fills up the state partition

ZubairLK, 2019-03-07:
Copied from https://github.com/balena-os/balena-os/issues/529

"We've seen this a couple of times, the fix being to remove the journal file and re-initialise the supervisor (if it broke).

Persistent logging is supposed to cut off after a certain size (8mb I believe) but we've seen cases of it taking all available state partition storage."

ZubairLK, 2019-03-07:
@agherzan  I think I've possibly root caused this one..

From the manual, https://www.freedesktop.org/software/systemd/man/systemd-journald.service.html
`If the daemon is stopped uncleanly, or if the files are found to be corrupted, they are renamed using the ".journal~" suffix, and systemd-journald starts writing to a new file`

`systemd-journald will automatically remove the oldest archived journal files to limit disk use.`

From some log snippets of devices where this happened, there are some corrupted file in `/var/log/journal`.
I'm guessing those weren't vacuumed.
Should have run `journalctl --verify` on those devices.. 

I'll try some unclean restarts on a device to make some corrupt journal entries and see if I can reproduce

ZubairLK, 2019-03-07:
hmm. might have been wrong.
on v2.26 tx2, journald only keeps 2 corrupted files while maintaining the max 8mb limit.
```
root@69b05b0:/var/log/journal/e9b03e9cfe09481792cc803c5c5ba80a# du -hs *
1.1M    system.journal
1.1M    system@0005838075c997fe-737073b095c193a7.journal~
1.1M    system@0005838099769c36-1d7a0d7a05a2b543.journal~
1.1M    system@43a9918ebf99427ca3ad8135c4fd7696-0000000000000001-0005838075a4e0b7.journal
1.1M    system@43a9918ebf99427ca3ad8135c4fd7696-000000000000050d-0005838075c78369.journal
1.1M    system@e0c8e327b27043c3a4b0dd76538e0ac5-0000000000000523-0005838068e71e6c.journal
1.1M    system@e25839f7043d4aa7a8f0888405e30153-0000000000000001-000583809964125b.journal
1.1M    system@e25839f7043d4aa7a8f0888405e30153-0000000000000522-000583809974f0ea.journal
root@69b05b0:/var/log/journal/e9b03e9cfe09481792cc803c5c5ba80a#
```

ZubairLK, 2019-03-07:
Tried many funky ways to make this happen. I can't seem to replicate this.

I tried dd-ing urandom into various journal files. The systemd-journal vacuum works and keeps logs 8.1Megs. occasionally I can get the folder to go a bit above 8.1megs. 

e.g. dd 500k or 900k into one/many of the journal files. But journald does hoover up the corrupted files.
```
root@62d0b9a:/var/log/journal/c84fbc2a7e40498e85d1ccf198799a02# du -hs *
1.1M    system.journal
1.1M    system@000583829eaa0f6e-97beb64069da1bda.journal~
1.1M    system@000583829ed10250-8c597080172e8f76.journal~
1.1M    system@43dc935dd94e44b4b929635381e70ead-0000000000000201-000583829ef279a3.journal
501K    system@54280a3ba6f24f7ab35be6723d4dd531-000000000001af2e-00058384c6f15f4d.journal
501K    system@54280a3ba6f24f7ab35be6723d4dd531-000000000001ba80-00058384c702f305.journal
501K    system@54280a3ba6f24f7ab35be6723d4dd531-000000000001c5c1-00058384c7147144.journal
501K    system@54280a3ba6f24f7ab35be6723d4dd531-000000000001d101-00058384c72d770e.journal
501K    system@54280a3ba6f24f7ab35be6723d4dd531-000000000001dc40-00058384c73f6d2c.journal
501K    system@54280a3ba6f24f7ab35be6723d4dd531-000000000001e77c-00058384c751d7f0.journal
501K    system@54280a3ba6f24f7ab35be6723d4dd531-000000000001f2c4-00058384c763adff.journal
501K    system@54280a3ba6f24f7ab35be6723d4dd531-000000000001fe00-00058384c775e938.journal
901K    system@54280a3ba6f24f7ab35be6723d4dd531-000000000002082b-00058384c936cc79.journal
root@62d0b9a:/var/log/journal/c84fbc2a7e40498e85d1ccf198799a02# du -hs ../
8.9M    ../
root@62d0b9a:/var/log/journal/c84fbc2a7e40498e85d1ccf198799a02# journalctl --verify
Journal file /var/log/journal/c84fbc2a7e40498e85d1ccf198799a02/system@54280a3ba6f24f7ab35be6723d4dd531-000000000001af2e-00058384c6f15f4d.journal corrupted, ignoring file.
PASS: /var/log/journal/c84fbc2a7e40498e85d1ccf198799a02/system@000583829ed10250-8c597080172e8f76.journal~                                      
PASS: /var/log/journal/c84fbc2a7e40498e85d1ccf198799a02/system.journal                                                                         
PASS: /var/log/journal/c84fbc2a7e40498e85d1ccf198799a02/system@000583829eaa0f6e-97beb64069da1bda.journal~                                      
PASS: /var/log/journal/c84fbc2a7e40498e85d1ccf198799a02/system@43dc935dd94e44b4b929635381e70ead-0000000000000201-000583829ef279a3.journal      
root@62d0b9a:/var/log/journal/c84fbc2a7e40498e85d1ccf198799a02# journalctl --list-boots
Journal file /var/log/journal/c84fbc2a7e40498e85d1ccf198799a02/system@54280a3ba6f24f7ab35be6723d4dd531-000000000001af2e-00058384c6f15f4d.journal corrupted, ignoring file.
 0 94c91b01055e4e619568035dfd9fd575 Thu 2019-03-07 15:18:38 UTC—Thu 2019-03-07 18:03:35 UTC
root@62d0b9a:/var/log/journal/c84fbc2a7e40498e85d1ccf198799a02# 
```

The journal rotates correctly..

ZubairLK, 2019-03-14:
Could be hitting https://github.com/systemd/systemd/issues/1017

ZubairLK, 2019-03-14:
We've seen this issue on the upboard which is running pyro i.e. systemd 2.32

majorz, 2019-07-31:
We hit this issue again on current in use systemd 239.

User was having malfunctioning POE and a Fin device was power cycled often. Persistent logging was enabled and the journal folder was twice the size - 16M instead of max 8M. Running `journalctl --verify` towards the recovered journal folder showed corruptions.

Additionally the /var/lib/NetworkManager folder was filled with many temp timestamp files - 2M in size. The combination of those filled the state partition.

On boot the system logs displayed:
```
[   14.609574] systemd-journald[478]: Received request to flush runtime journal from PID 1
[   14.655858] systemd-journald[478]: Failed to open system journal: No space left on device
[   14.667264] systemd-journald[478]: Failed to open system journal: No space left on device
[   14.677752] systemd-journald[478]: Failed to open system journal: No space left on device
[   14.689058] systemd-journald[478]: Failed to open system journal: No space left on device
[   14.699581] systemd-journald[478]: Failed to open system journal: No space left on device
[   14.710294] systemd-journald[478]: Failed to open system journal: No space left on device
[   14.720882] systemd-journald[478]: Failed to open system journal: No space left on device
...
```

This can be reproduced artificially - change the limit to 16M with enabled persistent logging. Let the journal folder be filled to that size. Then add a big file somewhere on the state partition to fill the rest of the free space (dd with a huge size does that leaving 0 bytes free). Return the limit to 8M and power cycle. You should see the above messages in the system logs. I did not test with WiFi connection only, but I suspect the device may not come online if this is so.

petrosagg, 2019-07-31:
That's good news @majorz. We should add this cleanup logic on boot to avoid this issue in the future

agherzan, 2019-07-31:
And we will for the next release.

commit 8284da08 (2019-08-08): systemd: Vacuum journald on boot

We have found a rare corner case bug where the journal bloats beyond
its limit and fills the state partition. Triggering a vacuum on reboot
helps a bit in case the device is restarted to recover its function.

Fixes #1423

Change-type: patch
Changelog-entry: Workaround for a cornercase bug in PersistentLogging where journalctl filled the state partition. Vacuum the journal on boot.
PR #1610 (2019-08-08): systemd: Vacuum journald on boot
We have found a rare corner case bug where the journal bloats beyond
its limit and fills the state partition. Triggering a vacuum on reboot
helps a bit in case the device is restarted to recover its function.

Fixes #1423

Change-type: patch
Changelog-entry: Workaround for a cornercase bug in PersistentLogging where journalctl filled the state partition. Vacuum the journal on boot.

---
### Contributor checklist

- [ ] Changes have been tested
- [ ] `Change-type` present on at least one commit
- [ ] `Signed-off-by` is present
- [ ] The PR complies with the [Ope [... cut]
closed 2019-08-14 by agherzan