balena-os/meta-balena #1452: balenaOS on desktop PC: Error: Can't have overlapping partitions

pdcastro, 2019-03-28:
User attempted to provision a 64-bit x86 desktop PC with an Intel NUC image and got the error: 
```
CP437: Invalid argument
fsck.fat 4.1 (2017-01-24)
/dev/disk/by-label/resin-boot: 286 files, 1505/20431 clusters
resin-rootA: clean, 5308/39936 files, 246429/319488 blocks
resin-rootB: clean, 11/39936 files, 15349/319488 blocks
resin-state: clean, 11/5136 files, 1849/20480 blocks
Error: Can't have overlapping partitions.
```
The user believes that the error happened _"in the second boot, on the first boot I do get the kind of old school greenish balena logo"._

@zvin preliminarily identified the issue as likely to come from `parted` in this code block:
https://github.com/balena-os/meta-balena/blob/v2.31.5/meta-resin-common/recipes-core/initrdscripts/files/resindataexpander#L37-L40

sako0938, 2019-03-29:
Something I also thought was odd with using the NUC image on my desktop computer was that it decided to repartition my ssd after booting from the MicroSD card. I would have expected Balena to expand the filesystem on the MicroSD card, like it would on a Raspberry Pi. If I wanted Balena running from my SSD I would have just flashed the img file onto the hard drive and booted from that, like I have done with the Pi3.

floion, 2019-03-29:
@sako0938 the Intel NUC image is an installer. If you used balena-cloud, in the instructions there is:

> Warning! This will also completely erase internal storage media, so please make a backup first.

sako0938, 2019-03-29:
I guess I missed that then :( After using Balena cloud for a few different arch types I assumed this one work the same and I glossed over that message I guess.

sako0938, 2019-03-29:
As a thought, maybe this message belongs in the big red box that for the Raspberry Pi Zero W says "The Raspberry Pi Zero W is not capable of connecting to 5GHz WiFi networks unless you use an external WiFi adapter that supports it."

ZubairLK, 2019-03-29:
@sako0938 thank-you for the feedback. I'll make an issue to fix this in the UI

ZubairLK, 2019-03-29:
@sako0938 See https://github.com/balena-io/resin-ui/issues/2710

pdcastro, 2019-04-01:
@ZubairLK, unless I misunderstood it, the main issue here is that provisioning a PC with the Intel NUC image failed with "Error: Can't have overlapping partitions." As per issue description, Alexis even identified the lines of code where the problem is probably coming from (`parted` command lines).

The UI change is an additional issue because the user wasn't expecting that the attempt to install balenaOS on the PC would repartition the internal disk - or at least not without the installer prompting the user to confirm before proceeding.

So I think this issue shouldn't be closed just yet...

sako0938, 2019-04-01:
Is there anything else I can do to help with this? Tonight I can get view of the partition layout that it produced. Not an expert on parted CLI by any means, but those lines look simple enough and they must work for NUCs?

sako0938, 2019-04-05:
@ZubairLK balena-io/resin-ui#2710 Is a bad link and there is no repo that I can find with that name, any ideas?

imrehg, 2019-04-06:
@sako0938 sorry, it's not a bad link, it's a private repository, and it's not visible to people outside of the balena org.

And that's also not the right place to make changes, nor is here @pdcastro @ZubairLK , the instructions are coming from the modules / device types used. For example for Intel, the device type setting the messages is here:

https://github.com/balena-os/balena-intel/blob/5c0b3782e79ddbd1c756dab3dd37dfbf549e5468/genericx86-64.coffee#L30-L35

And that relevant `instructions.FLASHER_WARNING` is coming from here:

https://github.com/balena-io-modules/resin-device-types/blob/3888375f38000b38dc7842b2540b3db3dc753d43/common/instructions.coffee#L120

Thus the issue and changes needs to be in https://github.com/balena-io-modules/resin-device-types/
Probably could use some help for a rewrite from the product team, cc @shaunmulligan

ZubairLK, 2019-04-08:
@sako0938 apologies. that link is a private repo. I didn't realise that.

And looks like due to the discussion around UI/UX, we missed the original issue.

```
Error: Can't have overlapping partitions.
```
@floion @spanceac have you guys seen this error on any flasher images? This guy is trying a nuc image on a desktop/server machine.

Side question: Do you have the USB/SD card with the flasher image still plugged in?
for flasher images, the flow usually is
a) flash flasher image onto usb/sd card
b) plug into powered off device
c) power on
d) flasher image will boot (or select that as usb/sd boot in bios etc)
e) the image triggers a shutdown by itself.
f) HAVE to remove the usb/sd card
g) Then reboot.

sako0938, 2019-04-08:
Not 100% sure I removed the installation medium on reboot, though pretty sure it didnt boot from it the second time. 
One thing I noticed was having a keyboard installed, if I pressed any key during the bootup stages it caused some kind of signal that doesnt get caught.

floion, 2019-04-09:
@ZubairLK this should be fixed by https://github.com/balena-os/meta-balena/pull/1379
@sako0938 what OS version did you try with?

sako0938, 2019-04-09:
@floion I used the latest Balena NUC image at the time, though I deleted the devices so all I know is the latest is: V2.31.2
Burned images using BalenaEtcher running on my Linux machine.

floion, 2019-05-17:
The last incarnation of this bug will be fixed by https://github.com/balena-os/meta-balena/pull/1506

closed 2019-05-17 by agherzan