balena-os/meta-balena #1461: Improve resilience of networking when user application can mess with things

imrehg, 2019-04-03:
Had a case when user application might have taken down the `resin-dns` interface due to incorrect use of udev + host dbus + udev rules in the cotnainer. 

This should not be possible, there should be ways to restore the interfaces that are taken down, for example.

ZubairLK, 2019-04-03:
log snippet from affected device.

```
root@8908936:~# journalctl   | grep resin-dns
Apr 03 08:04:23 localhost systemd-udevd[771]: Could not generate persistent MAC address for resin-dns: No such file or directory
Apr 03 08:04:23 8908936 avahi-daemon[788]: Joining mDNS multicast group on interface resin-dns.IPv6 with address fe80::4039:c3ff:fe55:55f1.
Apr 03 08:04:23 8908936 avahi-daemon[788]: New relevant interface resin-dns.IPv6 for mDNS.
Apr 03 08:04:23 8908936 avahi-daemon[788]: Joining mDNS multicast group on interface resin-dns.IPv4 with address 10.114.102.1.
Apr 03 08:04:23 8908936 avahi-daemon[788]: New relevant interface resin-dns.IPv4 for mDNS.
Apr 03 08:04:23 8908936 avahi-daemon[788]: Registering new address record for fe80::4039:c3ff:fe55:55f1 on resin-dns.*.
Apr 03 08:04:23 8908936 avahi-daemon[788]: Registering new address record for 10.114.102.1 on resin-dns.IPv4.
Apr 03 08:04:25 8908936 NetworkManager[787]: <info>  [1554278665.1923] manager: (resin-dns): new Bridge device (/org/freedesktop/NetworkManager/Devices/3)
Apr 03 08:04:48 8908936 systemd-udevd[1305]: Could not generate persistent MAC address for resin-dns: No such file or directory
Apr 03 08:04:53 8908936 avahi-daemon[788]: Withdrawing address record for 10.114.102.1 on resin-dns.
Apr 03 08:04:53 8908936 avahi-daemon[788]: Leaving mDNS multicast group on interface resin-dns.IPv4 with address 10.114.102.1.
Apr 03 08:04:53 8908936 avahi-daemon[788]: Interface resin-dns.IPv4 no longer relevant for mDNS.
Apr 03 08:04:53 8908936 avahi-daemon[788]: Interface resin-dns.IPv6 no longer relevant for mDNS.
Apr 03 08:04:53 8908936 avahi-daemon[788]: Leaving mDNS multicast group on interface resin-dns.IPv6 with address fe80::4039:c3ff:fe55:55f1.
Apr 03 08:04:53 8908936 avahi-daemon[788]: Withdrawing address record for fe80::4039:c3ff:fe55:55f1 on resin-dns.
root@8908936:~# 
```

That udevd line is suspicious. when looking at the full log, it appeared after some balena logs so it is probably from inside the container.

Nothing in the OS restarts resin-dns if its down. We probably should have something check that and attempt to bring the interface up.

ZubairLK, 2019-04-03:
currently we have distributed watchdogs on each service. which is necessary for things like balena, supervisor, vpn etc.

But I'm thinking we might benefit from one service monitoring other things as well such as this interface.
Needs an arch call I guess.

ZubairLK, 2019-04-03:
We might be hitting https://github.com/systemd/systemd/issues/3374

ZubairLK, 2019-04-03:
`resin-dns` is not managed by networkmanager as it is manually setup. Can't use NM dispatcher scripts as they stand.

resin-ci, 2019-04-09:
[zubairlk] This issue has attached support thread https://jel.ly.fish/#/support-thread~65b20bfc-347a-404e-b016-5c73070b8b8d

ZubairLK, 2019-04-10:
Traced this to `apt-get install -y connman` in a privileged app container (resin/raspberrypi3-node:6.9.2) with init system on.