balena-os/meta-balena #1463: Test timeinit-rtc and chrony behaviour when /dev/rtc appears after a container loads a kernel module

ZubairLK, 2019-04-04:
A container can load a driver that makes an rtc device node appear.

Test the behaviour of the OS (when it is disconnected from the internet so ntp sync isn't an issue)

Ideally, the appearance of /dev/rtc should trigger timeinit-rtc and chrony to sync up.

Not quite sure how chrony would read that time jump. IIRC chrony disables itself if timejumps like that are detected.

resin-ci, 2019-04-04:
[zubairlk] This issue has attached support thread https://jel.ly.fish/#/support-thread~c36b7a9b-1da9-44a4-b3a8-851623c5e9ba

ZubairLK, 2019-04-04:
Should be possible to manually replicate on the fin. Disable rtc module from loading. Disable wifi connection.
serial connection, old time, no /dev/rtc, insmod the rtc kernel module from a container

shaunmulligan, 2019-04-23:
@agherzan do you have a sense of how the OS will handle this case, where the RTC is only setup after user container starts up (modprobed,etc)?

agherzan, 2019-04-23:
@shaunmulligan, in that case, the init service will be skipped and the system time will not get updated. But even so, the RTC should be managed by chrony later when it appears (to be tested though).