balena-os/meta-balena #1466: do_kernel_resin_checkconfig failures should be errors instead of warnings

floion, 2019-04-05:
This will prevent us from deploying an OS with missing required kernel configs

agherzan, 2019-04-16:
@floion the issue with that is that because that mechanism is expected to work on various different kernel configs which might have or not those configs, we didn't want to block activating them on others.

floion, 2019-04-17:
I assume there are a number of these configs which are __absolutely necessary__ for running balena OS. These should be treated as errors in my view. The rest, which are optional can be warnings.

And even if we treat all as errors, if a particular board uses an older kernel which does not have some of these configs we can always unset them with an bbappend from the board directory to have the problem solved.