balena-os/meta-balena #1472: kernel-modules-headers and kernel-devsrc very slow

svenschwermer, 2019-04-15:
Hi,

The recipes `kernel-modules-headers` and `kernel-devsrc` have introduced a major compile time increase. Would it be possible to make those configureable (opt-in or opt-out, I don't really care)? As far as I can see, they're only needed if out-of-tree kernel modules need to be built.

Best regards,
Sven

agherzan, 2019-04-16:
Hi @svenschwermer . Are you building from scratch all the time? Isn't a shared sstate a viable solution?

svenschwermer, 2019-04-16:
> Hi @svenschwermer . Are you building from scratch all the time? Isn't a shared sstate a viable solution?

I am using an sstate, however, when developing on the kernel (recipe), the kernel frequently needs to be rebuilt and therefore those steps are executed very often.

svenschwermer, 2019-04-16:
I guess, I could just bbappend to `resin-image` and `resin-image-flasher` and
```
DEPENDS_remove = "kernel-modules-headers kernel-devsrc"
```

agherzan, 2019-04-16:
I was just about to recommend such a hack. Give it a try and see how it goes.

svenschwermer, 2019-04-16:
That works. Wouldn't it be more elegant though if meta-balena provided a class that out-of-tree kernel module recipes inherit from and automatically create then dependency?

agherzan, 2019-05-02:
I'm not sure I understand what you propose with the bbclass (maybe you can detail) but what I'm thinking is to add the depends conditionally based on a variable that you can overwrite in the build local.conf. Something like `OOT_KM_SUPPORT ?= "1"`.

svenschwermer, 2019-05-02:
> I'm not sure I understand what you propose with the bbclass (maybe you can detail) but what I'm thinking is to add the depends conditionally based on a variable that you can overwrite in the build local.conf. Something like `OOT_KM_SUPPORT ?= "1"`.

My idea was to create a class, say `oot-module.bbclass`, that would have the following contents:
```
DEPENDS = "kernel-modules-headers kernel-devsrc"
```
and then remove those dependencies from the image recipe(s). However, that would break existing OOT modules as they would have to inherit from that class. Also, this is not tested or anything. Your solution is probably better.

agherzan, 2019-05-02:
I see. Indeed we need to maintain the default behaviour which should be without any changes the build produces the kernel headers and src packages. Do you feel like testing a pushing a PR for this?

closed 2019-06-18 by svenschwermer