balena-os/meta-balena #1473: NetworkManager: failed 4-way handshake results in no connection retries

xginn8, 2019-04-15:
This issue occurs when a router and a device are booting at roughly the same time. The failed 4-way handshake is not retried, and NetworkManager considers the key invalid.

logs:

```sh
Apr 04 23:37:24 developmentpi NetworkManager[709]: <info>  [1554421044.4818] device (wlan0): Activation: (wifi) connection 'resin-wifi-01' has security, and secrets exist.  No new secrets needed.
Apr 04 23:37:24 developmentpi NetworkManager[709]: <info>  [1554421044.4821] Config: added 'ssid' value 'booster-truck-ca-069'
Apr 04 23:37:24 developmentpi NetworkManager[709]: <info>  [1554421044.4823] Config: added 'scan_ssid' value '1'
Apr 04 23:37:24 developmentpi NetworkManager[709]: <info>  [1554421044.4824] Config: added 'bgscan' value 'simple:30:-80:86400'
Apr 04 23:37:24 developmentpi NetworkManager[709]: <info>  [1554421044.4827] Config: added 'key_mgmt' value 'WPA-PSK'
Apr 04 23:37:24 developmentpi NetworkManager[709]: <info>  [1554421044.4828] Config: added 'auth_alg' value 'OPEN'
Apr 04 23:37:24 developmentpi NetworkManager[709]: <info>  [1554421044.4830] Config: added 'psk' value '<hidden>'
Apr 04 23:37:24 developmentpi NetworkManager[709]: <info>  [1554421044.6282] device (wlan0): supplicant interface state: inactive -> scanning
Apr 04 23:37:27 developmentpi NetworkManager[709]: <info>  [1554421047.4374] device (wlan0): supplicant interface state: scanning -> associating
Apr 04 23:37:27 developmentpi NetworkManager[709]: <info>  [1554421047.6069] device (wlan0): supplicant interface state: associating -> 4-way handshake
Apr 04 23:37:27 developmentpi NetworkManager[709]: <info>  [1554421047.6132] device (wlan0): supplicant interface state: 4-way handshake -> disconnected
Apr 04 23:37:27 developmentpi NetworkManager[709]: <info>  [1554421047.6157] device (wlan0): Activation: (wifi) disconnected during association, asking for new key
Apr 04 23:37:27 developmentpi NetworkManager[709]: <info>  [1554421047.6165] device (wlan0): state change: config -> need-auth (reason 'supplicant-disconnect', sys-iface-state: 'managed')
Apr 04 23:37:27 developmentpi NetworkManager[709]: <warn>  [1554421047.6237] device (wlan0): No agents were available for this request.
Apr 04 23:37:27 developmentpi NetworkManager[709]: <info>  [1554421047.6242] device (wlan0): state change: need-auth -> failed (reason 'no-secrets', sys-iface-state: 'managed')
Apr 04 23:37:27 developmentpi NetworkManager[709]: <info>  [1554421047.6268] manager: NetworkManager state is now CONNECTED_LOCAL
Apr 04 23:37:27 developmentpi NetworkManager[709]: <warn>  [1554421047.6304] device (wlan0): Activation: failed for connection 'resin-wifi-01'
Apr 04 23:37:27 developmentpi NetworkManager[709]: <info>  [1554421047.6321] device (wlan0): state change: failed -> disconnected (reason 'none', sys-iface-state: 'managed')
Apr 04 23:37:28 developmentpi NetworkManager[709]: <info>  [1554421048.1699] device (wlan0): set-hw-addr: set MAC address to 52:5E:DA:E0:55:C6 (scanning)
Apr 04 23:37:28 developmentpi NetworkManager[709]: <info>  [1554421048.1869] device (wlan0): supplicant interface state: disconnected -> inactive
Apr 04 23:37:28 developmentpi NetworkManager[709]: <info>  [1554421048.1983] device (wlan0): supplicant interface state: inactive -> disabled
Apr 04 23:37:28 developmentpi NetworkManager[709]: <info>  [1554421048.1987] device (wlan0): supplicant interface state: disabled -> inactive
Apr 04 23:42:46 developmentpi NetworkManager[709]: <info>  [1554421366.5687] device (wlan0): set-hw-addr: set MAC address to 4E:1B:D5:8E:73:18 (scanning)
Apr 04 23:42:46 developmentpi NetworkManager[709]: <info>  [1554421366.6004] device (wlan0): supplicant interface state: inactive -> disabled
Apr 04 23:42:46 developmentpi NetworkManager[709]: <info>  [1554421366.6008] device (wlan0): supplicant interface state: disabled -> inactive
Apr 04 23:48:02 developmentpi NetworkManager[709]: <info>  [1554421682.5636] device (wlan0): set-hw-addr: set MAC address to 8E:A4:B2:98:27:4D (scanning)
Apr 04 23:48:02 developmentpi NetworkManager[709]: <info>  [1554421682.5952] device (wlan0): supplicant interface state: inactive -> disabled
Apr 04 23:48:02 developmentpi NetworkManager[709]: <info>  [1554421682.5960] device (wlan0): supplicant interface state: disabled -> inactive
```

resin-ci, 2019-04-15:
[xginn8] This issue has attached support thread https://jel.ly.fish/#/support-thread~c7de7044-4d5c-4e7c-ad3d-3fef57adea3b

resin-ci, 2019-04-20:
[pdcastro] This issue has attached support thread https://jel.ly.fish/#/support-thread~68e2c196-1449-4611-8229-9b56854ac58f