balena-os/meta-balena #1486: User want to use wifi dongle with RTL8188CUS chipset

imrehg, 2019-04-29:
For example, EDUP EP-N8508GS wifi dongle.

imrehg, 2019-04-29:
Also came up in forums https://forums.balena.io/t/raspberry-pi-1-and-usb-wifi/6081

smilykoch, 2019-06-11:
+1 on this request. Support for RTL8188CUS would be awesome!

valtteric, 2019-07-16:
RTL8188CUS and RTL8188EUS are used in many mass produced (== high availability and cheap prices) wifi dongles for example by Edimax, TP-Link and D-Link. In my opinion supporting these should be prioritized really high.

opichals, 2019-08-27:
+1 as @valtteric noted here is a list of devices using this chipset https://cateee.net/lkddb/web-lkddb/RTL8192CU.html

FalconFour, 2020-06-12:
This was a deal breaker for me, as the EW-7811Un WiFi dongle I bought for my project (saying "Ideal for Raspberry Pi" on the box) doesn't work with Balena OS. Sad that it's gone a whole year without any meaningful progress towards fixing. Seems like nobody cares. Oh well, ditto! 👍