balena-os/meta-balena #1488: fsck runs on flasher images (it should not run when flashing)

floion, 2019-05-01:
usb-storage 2-1.3:1.0: USB Mass Storage device detected
scsi host0: usb-storage 2-1.3:1.0
usb 2-1.2.1: new full-speed USB device number 6 using ci_hdrc
starting version 237
usb 2-1.2.2: new full-speed USB device number 7 using ci_hdrc
hid-generic 0003:0DB5:0160.0001: hiddev0,hidraw0: USB HID v1.11 Device [Access IS NFC Smart Module] on usb-ci_hdrc.1-1.2.2/input1
CP437: Invalid argument
fsck.fat 4.1 (2017-01-24)
/dev/disk/by-label/resin-boot: 17 files, 3129/20431 clusters
resin-rootA: clean, 4165/39936 files, 190947/319488 blocks
resin-rootB: clean, 11/39936 files, 15349/319488 blocks
scsi 0:0:0:0: Direct-Access     Generic  STORAGE DEVICE   1532 PQ: 0 ANSI: 6
sd 0:0:0:0: Attached scsi generic sg0 type 0
resin-state: clean, 11/5136 files, 1849/20480 blocks
 mmcblk1: p1 p2 p3 p4 < p5 p6 >
sd 0:0:0:0: [sda] 62333952 512-byte logical blocks: (31.9 GB/29.7 GiB)
sd 0:0:0:0: [sda] Write Protect is off
sd 0:0:0:0: [sda] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
 sda: sda1 sda2 sda3 sda4 < sda5 sda6 >
sd 0:0:0:0: [sda] Attached SCSI removable disk
blk_update_request: I/O error, dev mmcblk1rpmb, sector 0
blk_update_request: I/O error, dev mmcblk1rpmb, sector 0

floion, 2019-05-01:
Created PR for fixing this https://github.com/balena-os/meta-balena/pull/1487

floion, 2019-05-17:
Will be fixed by https://github.com/balena-os/meta-balena/pull/1506

agherzan, 2019-05-17:
It's not merged yet so let's keep the issue open.

commit b931fd27 (2019-05-15): initrdscripts: Use the flasher boot argument for flasher detection

Fixes #1488

Change-type: minor
Changelog-entry: Switch flasher detection in initramfs to flasher boot parameter
commit 960068dd (2019-05-16): initrdscripts: Use the flasher boot argument for flasher detection

Fixes #1488

Change-type: minor
Changelog-entry: Switch flasher detection in initramfs to flasher boot parameter
commit 93e3c728 (2019-05-18): initrdscripts: Use the flasher boot argument for flasher detection

Fixes #1488

Change-type: minor
Changelog-entry: Switch flasher detection in initramfs to flasher boot parameter
closed 2019-05-17 by agherzan