balena-os/meta-balena #1491: Have the ability to disable mDNS / Avahi

pdcastro, 2019-05-01:
Customers reports that Avahi / mDNS is not required in their production devices and actually gets flagged as a security risk / concern / deployment impediment. Have the ability to disable it, similarly to the desire of disabling/blocking the ssh server in #1382.

pdcastro, 2020-12-07:
I think this issue may be considered resolved since supervisor v11.9.2 / balenaOS v2.52.4:
* https://github.com/balena-io/balena-supervisor/blob/master/CHANGELOG.md#v1192
* https://github.com/balena-os/meta-balena/blob/master/CHANGELOG.md#v2524

@richbayliss pointed out ([flowdock thread](https://www.flowdock.com/app/rulemotion/pub/threads/oIERczZfDypP4rmDQg8oTnIRseg)) that the BALENA_HOST_DISCOVERABILITY configuration variable exists and may be set to 'false' to prevent the avahi service from running on the device:

* https://github.com/balena-io/balena-supervisor/blob/17cf330771b0fa6e8b7aa22a4a245e7e3679abf8/src/device-config.ts#L189-L192
* https://www.balena.io/docs/learn/manage/configuration/

closed 2020-12-07 by pdcastro

Dids, 2022-04-09:
@pdcastro @richbayliss Should `BALENA_HOST_DISCOVERABILITY` or `RESIN_HOST_DISCOVERABILITY` still be working?

Because I'm trying to run Avahi inside a container, but balenaOS keeps trying to start Avahi, which of course results in port binding issues.

Rebooting does not help, and so far I've tried both configuration variables, which both seem to do nothing.

Dids, 2022-04-09:
Got an update on the situation! CC: @CameronDiver 

Toggling the config var to `1` and then `0` will stop Avahi:
```
...
Apr 09 18:23:27 balena faa3d862f5f2[1576]: [info]    Applying target state
Apr 09 18:23:27 balena balena-supervisor[2365]: [info]    Applying target state
Apr 09 18:23:27 balena balena-supervisor[2365]: [event]   Event: Apply config change in progress {}
Apr 09 18:23:27 balena faa3d862f5f2[1576]: [event]   Event: Apply config change in progress {}
Apr 09 18:23:27 balena faa3d862f5f2[1576]: [info]    Setting host to undiscoverable
Apr 09 18:23:27 balena balena-supervisor[2365]: [info]    Setting host to undiscoverable
Apr 09 18:23:27 balena faa3d862f5f2[1576]: [event]   Event: Apply config change success {}
Apr 09 18:23:27 balena balena-supervisor[2365]: [event]   Event: Apply config change success {}
Apr 09 18:23:27 balena balena-supervisor[2365]: [debug]   Stopping systemd unit: avahi-daemon.service
Apr 09 18:23:27 balena faa3d862f5f2[1576]: [debug]   Stopping systemd unit: avahi-daemon.service
Apr 09 18:23:27 balena faa3d862f5f2[1576]: [debug]   Stopping systemd unit: avahi-daemon.socket
Apr 09 18:23:27 balena balena-supervisor[2365]: [debug]   Stopping systemd unit: avahi-daemon.socket
Apr 09 18:23:27 balena avahi-daemon[3696]: Got SIGTERM, quitting.
...
Apr 09 18:23:27 balena avahi-daemon[3696]: avahi-daemon 0.8 exiting.
```

However, it will _immediately_ start itself back up again, which break the entire point behind this option:
```
...
Apr 09 18:23:27 balena dbus-daemon[1468]: [system] Activating via systemd: service name='org.freedesktop.Avahi' unit='dbus-org.freedesktop.Avahi.service' requested by ':1.51' (uid=0 pid=3669 comm="/usr/sbin/smbd --foreground --no-process-group --l")
...
Apr 09 18:23:27 balena avahi-daemon[11208]: Process 3696 died: No such process; trying to remove PID file. (/run/avahi-daemon//pid)
Apr 09 18:23:27 balena avahi-daemon[11208]: Found user 'avahi' (UID 994) and group 'avahi' (GID 993).
Apr 09 18:23:27 balena avahi-daemon[11208]: Successfully dropped root privileges.
Apr 09 18:23:27 balena avahi-daemon[11208]: avahi-daemon 0.8 starting up.
Apr 09 18:23:27 balena dbus-daemon[1468]: [system] Successfully activated service 'org.freedesktop.Avahi'
Apr 09 18:23:27 balena avahi-daemon[11208]: Successfully called chroot().
Apr 09 18:23:27 balena avahi-daemon[11208]: Successfully dropped remaining capabilities.
Apr 09 18:23:27 balena avahi-daemon[11208]: Loading service file /services/ssh.service.
...
```

To recap, the logic [here](https://github.com/balena-os/balena-supervisor/blob/master/src/lib/avahi.ts) seems to work fine for stopping Avahi, however for some reason it _immediately_ restarts it and judging by the logs, the restart request/action is triggered while Avahi is still exiting.