balena-os/meta-balena #1500: Redis image on Raspberry Pi 3 Model B+

dhenard, 2019-05-09:
Downloaded and installed the latest OS version balenaOS 2.32.0+rev1.  Tried to install the official redis image to the Raspberry Pi 3B+ and it kept restarting the container.  I then moved it to a new application that had just the one redis service.  The same error was occurring.  The new application had a docker-compose file of 
`version: '2'
volumes:
  iot-data:
services:
  redis_db:
    image: armhf/redis:latest
    container_name: redis
    network_mode: host
    volumes:
      - 'iot-data:/data'
    ports:
      - "127.0.0.1:6379:6379"
`

I then downgraded the host OS version back to balenaOS 2.31.5+rev1 and it started right up it both the new application and the original application.  Is there a setting that needs to be changed to work with the Redis image on the new version of the balenaOS or is this a bug?

shaunmulligan, 2019-05-09:
Hey @dhenard I believe you are hitting this issue https://github.com/balena-io/balena-supervisor/issues/970 which is due to having an `EXPOSE` in the redis. This has already been fixed but hasn't been released in a new OS yet, should be in 2.33.

agherzan, 2019-05-10:
The fix is included now in the latest release.

closed 2019-05-10 by agherzan