balena-os/meta-balena #1522: redsocks does not appear to support ssl proxy connections

memory, 2019-05-22:
At least as far as I can determine, it is not possible to configure redsocks to connect to a proxy over https/tls.  It can connect to SSL sites via sending HTTP CONNECT commands to a proxy, but the connection between redsocks and the proxy itself is done in cleartext.

Steps to reproduce:

1. You will need a squid proxy configures to support SSL (note that the default squid3 package in ubuntu bionic does not: I recommend using the packages built at https://github.com/diladele/squid-ubuntu if you don't feel like recompiling squid from scratch)

A sample squid.conf with an ssl listener on port 443:

```
# ACLS
acl manager proto cache_object
acl localhost src 127.0.0.1/32
acl to_localhost dst 127.0.0.0/8
acl all src 0.0.0.0/0

# SAFE PORTS
acl SSL_ports  port 443
acl Safe_ports port 443
acl Safe_ports port 80          # http
acl Safe_ports port 21          # ftp
acl Safe_ports port 443         # https
acl Safe_ports port 70          # gopher
acl Safe_ports port 210         # wais
acl Safe_ports port 1025-65535  # unregistered ports
acl Safe_ports port 280         # http-mgmt
acl Safe_ports port 488         # gss-http
acl Safe_ports port 591         # filemaker
acl Safe_ports port 777         # multiling http
acl CONNECT method CONNECT
acl authme proxy_auth REQUIRED
always_direct allow all
always_direct allow CONNECT

# HTTP ACCESS
http_access deny !authme
http_access allow manager localhost
http_access deny manager
http_access deny !Safe_ports
http_access deny CONNECT !SSL_ports
http_access allow localhost
http_access allow all
#icp_access deny all
#htcp_access deny all

# ports
http_port 3128
https_port 443 tls-cert=/etc/squid/proxy.crt tls-key=/etc/squid/proxy.key options=NO_SSLv3,SINGLE_DH_USE,SINGLE_ECDH_USE tls-dh=/etc/squid/dhparam.pem

# CACHE CFG
cache_replacement_policy lru
cache_mem 256 MB
maximum_object_size_in_memory 2560 KB
cache_dir ufs /var/spool/squid 5120 16 256
maximum_object_size 1048576 KB

refresh_pattern ^ftp:           1440    20%     10080
refresh_pattern ^gopher:        1440    0%      1440
refresh_pattern (cgi-bin|\?)    0       0%      0
refresh_pattern .               0       20%     4320

#TIMEOUTS
connect_timeout 8 seconds
peer_connect_timeout 3 seconds

coredump_dir /var/spool/squid

#tls_outgoing_options cipher=ALL:!EXPORT:!EXPORT40:!EXPORT56:!aNULL:!LOW:!RC4:@STRENGTH
tls_outgoing_options cipher=HIGH:MEDIUM:!RC4:!aNULL:!eNULL:!LOW:!3DES:!MD5:!EXP:!PSK:!SRP:!DSS
```

Note that this configuration sets up a globally-accessible proxy: you will want to limit access by using `htpasswd` to create `/etc/squid/squid.pass`.  You will also need to create an openssl key and a self-signed cert in /etc/squid.

2. Configure redsocks to connect to your proxy on port 443:

```
base {
log_debug = off;
log_info = on;
log = "syslog:local7";
daemon = off;
redirector = iptables;
}
redsocks {
type = http-connect;
ip = X.X.X.X;  # <- add the IP address of your squid proxy
port = 443;
local_ip = 127.0.0.1;
local_port = 12345;
login = "myuser";  # <- user/pass should match your squid.pass file
password = "mypass";
}
```

The device will not come online in this configuration: tcpdump on your proxy will reveal that it is not attempting to negotiate a TLS connection, but is instead sending plaintext HTTP CONNECT commands to port 443.

If you configure redsocks to talk to the http_port on 3128/tcp, the device will successfully come online.

In contrast, you should be able to successfully use curl to proxy requests over TSL:

```
curl -v -U myuser:mypass -x https://testproxy.balena.io/ --proxy-insecure https://ifconfig.me
```

agherzan, 2019-05-22:
That is correct. redsocks doesn't support https and the author is not planning to add any features (https://github.com/darkk/redsocks/issues/128). There is a fork mentioned which we haven't played with until now but, checking the code, it does implement this functionality (https://github.com/semigodking/redsocks/blob/master/https-connect.c).

bbinet, 2019-05-23:
I'm also interested in moving to [Redsocks2](https://github.com/semigodking/redsocks) as it also has integrated [shadowsocks](http://shadowsocks.org) proxy support.

closed 2020-09-27 by memory