balena-os/meta-balena #1523: File system image name is `-img` instead of `.img`

imrehg, 2019-05-22:
The resulting disk image of the build for balenaOS seems to be `.resin-img`, and not a well-established file extension such as `.img`. Because of that, e.g. if the file is compressed in an archive, etcher cannot burn it, as it doesn't find any suitable image. For images downloaded from the dashboard, we already change this file extension to `.img` inside the compressed archive, would it make sense to have it named such at the output of the build, so that no manual renaming is needed when needing to share the file in a compressed form?

agherzan, 2019-05-22:
@imrehg this is part of the way yocto generates the final artefacts. I think we should address this at the level of deployment to s3 (https://github.com/balena-os/balena-yocto-scripts).

imrehg, 2019-05-24:
@agherzan it's sort of fair point, and I wasn't sure where to bring it up, might be easily `wontfix` or `invalid`. The thing is, from experience I've seen people just taking the output of their local yocto build and compressing, then sharing. Thus since it's a) not really Etcher's part to know about that extension which I admit, b) S3 might not be involved at all in practical use, then the only thing is: *for people to always remember to manually rename*, right? Not ideal, but indeed no obvious place where to fix it, if not in meta-balena (and how assets are generated, or what symlinks, etc are created).

Anyways, just a braindump, in case other people run into the same issue as well, not being able directly burn the generated image from a compressed file. Thanks!

agherzan, 2019-05-24:
I see. That is a fair argument. I'll think about but it most probably is a low priority.

alexgg, 2020-04-17:
Came up again https://www.flowdock.com/app/rulemotion/r-resinos/threads/gx_k0FXcb9zCrSDx2zl3bFK9JJl