balena-os/meta-balena #1525: docker in docker: docker-disk installs in host

siredmar, 2019-05-23:
Hi!
I'm trying to set up a build chain where balenaOs is built in a docker container.
What i basically do is the following:

1. while docker build: 
* install docker in the image

2. while docker run: 
* mount the hosts /var/run/docker.sock to use the hosts docker daemon (remember [PR #1417](https://github.com/balena-os/meta-balena/pull/1474))
* mount the checked out balena repository to /balena-build in the container
* trigger barys to start the yocto build

The process seems to work fine except for the docker-disk recepie.
The build process fails because of the missing resin-data.img that doesn't get installed within the containers /balena-build directory but on the **hosts /balena-build direcory** (which should not exist at all!)

Here are some logs:
```
NOTE: Running task 3371 of 3855 (/build-balena/balena-raspberrypi/build/../layers/meta-balena/meta-resin-common/recipes-containers/docker-disk/docker-disk.bb:do_compile)
NOTE: Running task 3422 of 3855 (/build-balena/balena-raspberrypi/build/../layers/meta-balena/meta-resin-common/recipes-core/images/resin-image.bb:do_image_hostapp_ext4)
NOTE: recipe docker-disk-balena_armv7hf-supervisor-v9.15.0-r0: task do_compile: Started
NOTE: recipe resin-image-1.0-r0: task do_image_hostapp_ext4: Started
ERROR: resin-image-1.0-r0 do_image_hostapp_ext4: Function failed: do_image_hostapp_ext4 (log file is located at /build-balena/balena-raspberrypi/build/tmp/work/raspberrypi3-poky-linux-gnueabi/resin-image/1.0-r0/temp/log.do_image_hostapp_ext4.315)
ERROR: Logfile of failure stored in: /build-balena/balena-raspberrypi/build/tmp/work/raspberrypi3-poky-linux-gnueabi/resin-image/1.0-r0/temp/log.do_image_hostapp_ext4.315
Log data follows:
| DEBUG: Executing python function set_image_size
| DEBUG: 208492.000000 = 208492 * 1.000000
| DEBUG: 319488.000000 = max(208492.000000, 319488)[319488.000000] + 1
| DEBUG: 319488.000000 = int(319488.000000)
| DEBUG: 319488 = aligned(319488)
| DEBUG: returning 319488
| DEBUG: Python function set_image_size finished
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['virtual:native:/build-balena/balena-raspberrypi/build/../layers/poky/meta/recipes-extended/pigz/pigz_2.4.bb:do_populate_sysroot', 'virtual:native:/build-balena/balena-raspberrypi/build/../layers/poky/meta/recipes-devtools/opkg/opkg_0.3.6.bb:do_populate_sysroot', '/build-balena/balena-raspberrypi/build/../layers/poky/meta/recipes-kernel/kmod/depmodwrapper-cross_1.0.bb:do_populate_sysroot', '/build-balena/balena-raspberrypi/build/../layers/meta-balena/meta-resin-common/recipes-containers/mkfs-hostapp-native/mkfs-hostapp-native.bb:do_populate_sysroot', '/build-balena/balena-raspberrypi/build/../layers/poky/meta/recipes-devtools/mklibs/mklibs-native_0.1.43.bb:do_populate_sysroot', 'virtual:native:/build-balena/balena-raspberrypi/build/../layers/poky/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', 'virtual:native:/build-balena/balena-raspberrypi/build/../layers/meta-openembedded/meta-oe/recipes-devtools/jq/jq_1.5.bb:do_populate_sysroot', 'virtual:native:/build-balena/balena-raspberrypi/build/../layers/poky/meta/recipes-devtools/opkg-utils/opkg-utils_0.3.6.bb:do_populate_sysroot', 'virtual:native:/build-balena/balena-raspberrypi/build/../layers/poky/meta/recipes-core/update-rc.d/update-rc.d_0.8.bb:do_populate_sysroot', 'virtual:native:/build-balena/balena-raspberrypi/build/../layers/poky/meta/recipes-core/coreutils/coreutils_8.30.bb:do_populate_sysroot', '/build-balena/balena-raspberrypi/build/../layers/poky/meta/recipes-devtools/qemu/qemuwrapper-cross_1.0.bb:do_populate_sysroot', '/build-balena/balena-raspberrypi/build/../layers/poky/meta/recipes-core/glibc/ldconfig-native_2.12.1.bb:do_populate_sysroot', 'virtual:native:/build-balena/balena-raspberrypi/build/../layers/poky/meta/recipes-devtools/makedevs/makedevs_1.0.1.bb:do_populate_sysroot', '/build-balena/balena-raspberrypi/build/../layers/poky/meta/recipes-core/glibc/cross-localedef-native_2.28.bb:do_populate_sysroot', 'virtual:native:/build-balena/balena-raspberrypi/build/../layers/poky/meta/recipes-devtools/prelink/prelink_git.bb:do_populate_sysroot', '/build-balena/balena-raspberrypi/build/../layers/meta-balena/meta-resin-common/recipes-devtools/kernel-modules-headers/kernel-modules-headers.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['pigz-native', 'opkg-native', 'depmodwrapper-cross', 'mkfs-hostapp-native', 'mklibs-native', 'pseudo-native', 'jq-native', 'opkg-utils-native', 'update-rc.d-native', 'coreutils-native', 'qemuwrapper-cross', 'ldconfig-native', 'makedevs-native', 'cross-localedef-native', 'prelink-native', 'kernel-modules-headers', 'xz-native', 'kmod-native', 'systemd-systemctl-native', 'zlib-native', 'quilt-native', 'libarchive-native', 'autoconf-native', 'pkgconfig-native', 'libsolv-native', 'automake-native', 'libtool-native', 'gnu-config-native', 'hostapp-update-native', 'e2fsprogs-native', 'balena-native', 'shadow-native', 'gettext-minimal-native', 'qemu-native', 'onig-native', 'texinfo-dummy-native', 'attr-native', 'nss-native', 'elfutils-native', 'binutils-native', 'linux-libc-headers', 'openssl', 'elfutils', 'linux-raspberrypi', 'util-linux', 'glibc', 'gcc-runtime', 'openssl-native', 'debianutils-native', 'python3-native', 'gtk-doc-native', 'bzip2-native', 'lzo-native', 'm4-native', 'rpm-native', 'ninja-native', 'expat-native', 'cmake-native', 'util-linux-native', 'go-native', 'pixman-native', 'glib-2.0-native', 'alsa-lib-native', 'dtc-native', 'sqlite3-native', 'nspr-native', 'flex-native', 'libtool', 'bzip2', 'zlib', 'cpio-native', 'ncurses', 'libxcrypt', 'opkg-utils', 'bash-completion', 'libgcc', 'readline-native', 'gdbm-native', 'dbus-native', 'file-native', 'popt-native', 'db-native', 're2c-native', 'curl-native', 'ncurses-native', 'util-macros-native', 'libpng-native', 'gettext-native', 'libffi-native', 'libpcre-native']
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function do_image_hostapp_ext4
| 0+0 records in
| 0+0 records out
| 0 bytes copied, 1.6891e-05 s, 0.0 kB/s
| File does not exist: /build-balena/balena-raspberrypi/build/tmp/work/raspberrypi3-poky-linux-gnueabi/resin-image/1.0-r0/deploy-resin-image-image-complete/resin-image-raspberrypi3-20190523091936.rootfs.docker
| WARNING: /build-balena/balena-raspberrypi/build/tmp/work/raspberrypi3-poky-linux-gnueabi/resin-image/1.0-r0/temp/run.do_image_hostapp_ext4.315:1 exit 1 from 'mkfs.hostapp-ext4 -t "/build-balena/balena-raspberrypi/build/tmp" -s "/build-balena/balena-raspberrypi/build/tmp/work/raspberrypi3-poky-linux-gnueabi/resin-image/1.0-r0/recipe-sysroot-native" -i /build-balena/balena-raspberrypi/build/tmp/work/raspberrypi3-poky-linux-gnueabi/resin-image/1.0-r0/deploy-resin-image-image-complete/resin-image-raspberrypi3-20190523091936.rootfs.docker -o /build-balena/balena-raspberrypi/build/tmp/work/raspberrypi3-poky-linux-gnueabi/resin-image/1.0-r0/deploy-resin-image-image-complete/resin-image-raspberrypi3-20190523091936.rootfs.hostapp-ext4'
| ERROR: Function failed: do_image_hostapp_ext4 (log file is located at /build-balena/balena-raspberrypi/build/tmp/work/raspberrypi3-poky-linux-gnueabi/resin-image/1.0-r0/temp/log.do_image_hostapp_ext4.315)
NOTE: recipe resin-image-1.0-r0: task do_image_hostapp_ext4: Failed
ERROR: Task (/build-balena/balena-raspberrypi/build/../layers/meta-balena/meta-resin-common/recipes-core/images/resin-image.bb:do_image_hostapp_ext4) failed with exit code '1'
NOTE: recipe docker-disk-balena_armv7hf-supervisor-v9.15.0-r0: task do_compile: Succeeded
NOTE: Tasks Summary: Attempted 3843 tasks of which 3841 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /build-balena/balena-raspberrypi/build/../layers/meta-balena/meta-resin-common/recipes-core/images/resin-image.bb:do_image_hostapp_ext4
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
[000000037][LOG]Build for raspberrypi3 failed. Check failed log in build/tmp/log/cooker/raspberrypi3 .
[000000037][LOG]If build for raspberrypi3 succeeded, final image should have been generated here:
[000000037][LOG]   build/tmp/deploy/images/raspberrypi3/resin-image-raspberrypi3.resinos-img
[000000037][LOG]Done.
[ERROR] failed to execute task "make-build-yocto:": exited with non-zero status code 2

```

When i look in my hosts root fs i find this path (that should not exist at all on the host):
```
[... cut ...]
So there is really something fishy. I noticed that you start a real docker daemon to create the supervisor(?) image. Why is that? I often use docker in docker mouting the docker.sock but never ever started a real docker daemon within a running container. Please see [do-not-use-docker-in-docker-for-ci](https://jpetazzo.github.io/2015/09/03/do-not-use-docker-in-docker-for-ci/)

What i expect is that that the supervisor resin-data.img gets deployed within the docker container and not on the hosts filesystem. But this way the build won't succeed because of the missing artifacts from the docker-disk steps.

agherzan, 2019-05-23:
@siredmar The build system doesn't only create docker image but it actually initializes the entire data partition for balena-engine runtime data/metadata. Which means in order to be able to do this cleanly, a docker in docker is needed. We do that in the CI as well and it works just fine (see https://github.com/balena-os/balena-yocto-scripts/blob/master/automation/prepare-and-start.sh) without any signs of corruption.

siredmar, 2019-05-24:
Thx for responding! I'm aware of your CI Dockerfile and this prepare-and-start script. I tried the exact versions and found out the docker daemon inside didn't came up. Don't have the logs saved but iirc it was something like my kernel does not support aufs. And something deprecated stuff... 

Currently i am testing a modified version of your files where the hosts docker.sock is mounted and another known working docker version is installed. I'll report back if i have news.

agherzan, 2019-05-24:
Indeed. building balenaOS currently has the aufs requirement for some boards. We are working to migrate to overlayfs for all our supported devices.

closed 2019-05-24 by agherzan

siredmar, 2019-05-27:
link: https://github.com/balena-os/balena-yocto-scripts/issues/140
FYI: i was giving up the idea of providing my own docker environment and now using your jenkins_build.sh script.