balena-os/meta-balena #1526: Allow configuring chrony from config.json to allow GPS as a time source

balena-ci, 2019-05-24:
[shaunmulligan]  modify chrony.conf to add a gps source server from GPSD

```
driftfile /var/lib/chrony/drift

allow

# set larger delay to allow the NMEA source to overlap with
# the other sources and avoid the falseticker status
refclock SHM 0 refid GPS precision 1e-1 offset 0.9999 delay 0.2
refclock SOCK /var/run/chrony.ttyACM0.sock refid PPS

```

agherzan, 2019-05-27:
@shaunmulligan `/var/run/chrony.ttyACM0.sock` is supposed to be managed by gpsd. Can you describe what is the expected user interface for this feature?

daghemo, 2020-03-26:
Any news on the issue?

I may be interested in it, since I'm playing with GNSS support within Quectel 3G/4G modules, but I also need to use other non common time references (e.g. a DSLR camera connected via USB). 
I can't see the support thread mentioned here above, but I think that the issue is related to https://forums.balena.io/t/gpsd-source-for-chrony/10057.

I know that I can also run a separate Chrony (or even another NTP) daemon within the container, but would it be better to have a correct clock within the host OS or the container itself? 
I suppose not both: the host OS or (XOR actually) the container should be given the rights to "keep" the time. 

Maybe I can try to pass an empy list to ntpServers as shown on https://www.balena.io/docs/reference/OS/time/ and just set some NTP servers within the container.
Also, I know for sure the running a privileged container I could even run "date --set=<something>" within the container itself to also change date and time on the host OS.

That is, playing with NTP (only) within the container should work, but... do you see any impact on the host OS, the Supervisor or whatever else?

From the other side, we may play with NTP on the host side for sure, but we are limited to NTP servers and RTC right now, if I'm not wrong. Adding support for SOCK and SHM would be greater, e.g. to operate a gpsd to obtain the correct time via GPS, but we should probably keep in mind that both SOCK and SHM are not shared by default from the host OS to the container!
That is, gpsd should run within the host OS (and I think this is not doable) or we should use a docker-compose.yml sharing the SOCK via bind mounting and the SHM via the ipc=host option.
Thanks to "ipc=host" web can then run gpsd within the container, or even write directly to the shared memory settings as required by the ntpd driver 28.

Maybe you can give the option so pass "refclock" lines just you do for udevRules, so everybody can chose, also deciding the preferred offset, precision and delay for the his/her refclocks.

Timple, 2021-06-15:
Maybe "Allow configuring chrony from config.json" would also be a proper title.
Because we need to adjust out-of-the ordinary things in the config as well (higher polling frequency).

Providing access to the full config would solve both our issues.