balena-os/meta-balena #1527: Improve IWLWIFI_FW_TOCLEAN and balena-linux-firmware.bbclass

ZubairLK, 2019-05-24:
I think it misses some firmware.

e.g. This is defined in the thud layer
```
inherit balena-linux-firmware

# Cleanup iwlwifi firmware files
IWLWIFI_PATH = "lib/firmware"
IWLWIFI_REGEX = "^iwlwifi-([0-9a-zA-Z-]+)-([0-9]+).ucode$"
IWLWIFI_FW_TOCLEAN ?= " \
    7260 \
    7265 \
    7265D \
    8000C \
    8265 \
"
IWLWIFI_FW_MIN_API[7260] = "17"
IWLWIFI_FW_MIN_API[7265] = "17"
IWLWIFI_FW_MIN_API[7265D] = "29"
IWLWIFI_FW_MIN_API[8000C] = "34"
IWLWIFI_FW_MIN_API[8265] = "34"
```

Note the existence of the following files in the rootfs: `iwlwifi-7260-7.ucode` `iwlwifi-7260-8.ucode` `iwlwifi-7260-9.ucode` even though `IWLWIFI_FW_MIN_API[7265] = "17"` is defined

```
bash-4.4# cat /etc/os-release 
ID="balena-os"
NAME="balenaOS"
VERSION="2.36.0+rev2"
VERSION_ID="2.36.0"
PRETTY_NAME="balenaOS 2.36.0+rev2"
MACHINE="genericx86-64"
VARIANT="Development"
VARIANT_ID="dev"
RESIN_BOARD_REV="6c6d325"
META_RESIN_REV="2fc8e3c"
SLUG="intel-nuc"
bash-4.4# ls
LICENCE.atheros_firmware      iwlwifi-6000-4.ucode     iwlwifi-9000-pu-b0-jf-b0-33.ucode
LICENCE.broadcom_bcm43xx      iwlwifi-6000g2a-5.ucode  iwlwifi-9000-pu-b0-jf-b0-34.ucode
LICENCE.ibt_firmware          iwlwifi-6000g2a-6.ucode  iwlwifi-9000-pu-b0-jf-b0-38.ucode
LICENCE.iwlwifi_firmware      iwlwifi-6000g2b-5.ucode  iwlwifi-9000-pu-b0-jf-b0-41.ucode
LICENCE.ralink-firmware.txt   iwlwifi-6000g2b-6.ucode  mrvl
LICENCE.rtlwifi_firmware.txt  iwlwifi-6050-4.ucode     rt2561.bin
ar7010.fw                     iwlwifi-6050-5.ucode     rt2561s.bin
ar7010_1_1.fw                 iwlwifi-7260-17.ucode    rt2661.bin
ar9271.fw                     iwlwifi-7260-7.ucode     rt2860.bin
ath10k                        iwlwifi-7260-8.ucode     rt2870.bin
ath9k_htc                     iwlwifi-7260-9.ucode     rt3070.bin
brcm                          iwlwifi-7265-17.ucode    rt3071.bin
htc_7010.fw                   iwlwifi-7265-8.ucode     rt3090.bin
htc_9271.fw                   iwlwifi-7265-9.ucode     rt3290.bin
intel                         iwlwifi-7265D-29.ucode   rt73.bin
iwlwifi-135-6.ucode           iwlwifi-8000C-34.ucode   rtl_bt
iwlwifi-3160-7.ucode          iwlwifi-8000C-36.ucode   rtl_nic
iwlwifi-3160-8.ucode          iwlwifi-8265-34.ucode    rtlwifi
iwlwifi-3160-9.ucode          iwlwifi-8265-36.ucode
bash-4.4# 
```

agherzan, 2019-05-24:
@ZubairLK is this reproducible on any current master board? Otherwise please tell me what board have you reproduced this on and what revision.

ZubairLK, 2019-05-24:
@agherzan I ran the latest docker file to check this.
`docker run --rm -i -t resin/resinos:2.36.0_rev2.dev-intel-nuc /bin/bash`

```
bash-4.4# cat /etc/os-release 
ID="balena-os"
NAME="balenaOS"
VERSION="2.36.0+rev2"
VERSION_ID="2.36.0"
PRETTY_NAME="balenaOS 2.36.0+rev2"
MACHINE="genericx86-64"
VARIANT="Development"
VARIANT_ID="dev"
RESIN_BOARD_REV="6c6d325"
META_RESIN_REV="2fc8e3c"
SLUG="intel-nuc"
```

commit 738e0c29 (2019-05-24): balena-linux-firmware: Firmware clean various improvements

This adds an easier to debug logging that shows both removed and kept
firmwares. For example:

 Cleaning iwlwifi firmware version 8000C to minimum API version 34.
   Removing iwlwifi-8000C-13.ucode.
   Removing iwlwifi-8000C-22.ucode.
   Removing iwlwifi-8000C-21.ucode.
   Removing iwlwifi-8000C-31.ucode.
   Removing iwlwifi-8000C-16.ucode.
   Removing iwlwifi-8000C-27.ucode.
   Leaving iwlwifi-8000C-34.ucode in place.
   Leaving iwlwifi-8000C-36.ucode in place.

Also, this commit improves the version comparison which was comparing
st [... cut]
commit a415867a (2019-05-24): balena-linux-firmware: Firmware clean various improvements

This adds an easier to debug logging that shows both removed and kept
firmwares. For example:

 Cleaning iwlwifi firmware version 8000C to minimum API version 34.
   Removing iwlwifi-8000C-13.ucode.
   Removing iwlwifi-8000C-22.ucode.
   Removing iwlwifi-8000C-21.ucode.
   Removing iwlwifi-8000C-31.ucode.
   Removing iwlwifi-8000C-16.ucode.
   Removing iwlwifi-8000C-27.ucode.
   Leaving iwlwifi-8000C-34.ucode in place.
   Leaving iwlwifi-8000C-36.ucode in place.

Also, this commit improves the version comparison which was comparing
st [... cut]
commit 6500c7f0 (2019-05-24): balena-linux-firmware: Firmware clean various improvements

This adds an easier to debug logging that shows both removed and kept
firmwares. For example:

 Cleaning iwlwifi firmware version 8000C to minimum API version 34.
   Removing iwlwifi-8000C-13.ucode.
   Removing iwlwifi-8000C-22.ucode.
   Removing iwlwifi-8000C-21.ucode.
   Removing iwlwifi-8000C-31.ucode.
   Removing iwlwifi-8000C-16.ucode.
   Removing iwlwifi-8000C-27.ucode.
   Leaving iwlwifi-8000C-34.ucode in place.
   Leaving iwlwifi-8000C-36.ucode in place.

Also, this commit improves the version comparison which was comparing
st [... cut]
closed 2019-05-29 by agherzan