balena-os/meta-balena #1531: 2.36.0+rev1 seems to break MDNS support

hedss, 2019-05-24:
Trying to use a BoB with `.local` domain returns this in os-config:
```
May 24 14:27:35 05ad2b2 systemd[1]: Started OS configuration update service.
May 24 14:27:35 05ad2b2 os-config[3528]: Fetching service configuration from https://api.new.katapult.local/os/v1/config...
May 24 14:27:35 05ad2b2 os-config[3528]: sending message via: UDP(127.0.0.2:53)
May 24 14:27:35 05ad2b2 os-config[3528]: sending message via: UDP(127.0.0.2:53)
May 24 14:27:35 05ad2b2 os-config[3528]: https://api.new.katapult.local/os/v1/config: an error occurred trying to connect: no record found for name: api.new.katapult.local type: AAAA class: IN
May 24 14:27:36 05ad2b2 os-config[3528]: sending message via: UDP(127.0.0.2:53)
May 24 14:27:36 05ad2b2 os-config[3528]: sending message via: UDP(127.0.0.2:53)
May 24 14:27:37 05ad2b2 os-config[3528]: sending message via: UDP(127.0.0.2:53)
May 24 14:27:37 05ad2b2 os-config[3528]: sending message via: UDP(127.0.0.2:53)
```
and this from the Supervisor:
```
May 24 14:20:27 05ad2b2 resin-supervisor[2593]: Error reporting initial configuration, will retry { Error: getaddrinfo ENOTFOUND api.new.katapult.local api.new.katapult.local:443
May 24 14:20:27 05ad2b2 resin-supervisor[2593]:     at GetAddrInfoReqWrap.onlookup [as oncomplete] (dns.js:56:26)
May 24 14:20:27 05ad2b2 resin-supervisor[2593]:   cause:
May 24 14:20:27 05ad2b2 resin-supervisor[2593]:    { Error: getaddrinfo ENOTFOUND api.new.katapult.local api.new.katapult.local:443
May 24 14:20:27 05ad2b2 resin-supervisor[2593]:        at GetAddrInfoReqWrap.onlookup [as oncomplete] (dns.js:56:26)
May 24 14:20:27 05ad2b2 resin-supervisor[2593]:      errno: 'ENOTFOUND',
May 24 14:20:27 05ad2b2 resin-supervisor[2593]:      code: 'ENOTFOUND',
May 24 14:20:27 05ad2b2 resin-supervisor[2593]:      syscall: 'getaddrinfo',
May 24 14:20:27 05ad2b2 resin-supervisor[2593]:      hostname: 'api.new.katapult.local',
May 24 14:20:27 05ad2b2 resin-supervisor[2593]:      host: 'api.new.katapult.local',
May 24 14:20:27 05ad2b2 resin-supervisor[2593]:      port: 443 },
May 24 14:20:27 05ad2b2 resin-supervisor[2593]:   isOperational: true,
May 24 14:20:27 05ad2b2 resin-supervisor[2593]:   errno: 'ENOTFOUND',
May 24 14:20:27 05ad2b2 resin-supervisor[2593]:   code: 'ENOTFOUND',
May 24 14:20:27 05ad2b2 resin-supervisor[2593]:   syscall: 'getaddrinfo',
May 24 14:20:27 05ad2b2 resin-supervisor[2593]:   hostname: 'api.new.katapult.local',
May 24 14:20:27 05ad2b2 resin-supervisor[2593]:   host: 'api.new.katapult.local',
May 24 14:20:27 05ad2b2 resin-supervisor[2593]:   port: 443 }
```
It should be noted that `ping api.new.katapult.local` correctly resolves the address, and that `/etc/nsswitch.conf` does include a reference to `mdns4_minimal`. So DNS resolution *is* working, but for some reason neither `os-config` or the Supervisor are able to use it (device will registry with API, will download app, but obviously won't connect to the VPN as there's no VPN config as os-config can't resolve the address).

agherzan, 2019-05-24:
I tried to replicate this on a RaspberryPi 3 but I don't seem to be able to:

```
[andrei@fluxx ~]$ ssh -p 22222 root@10.100.100.102 ping balena.local
PING balena.local (10.100.100.103): 56 data bytes
64 bytes from 10.100.100.103: seq=0 ttl=64 time=6.455 ms
64 bytes from 10.100.100.103: seq=1 ttl=64 time=17.893 ms
64 bytes from 10.100.100.103: seq=2 ttl=64 time=3.309 ms
[andrei@fluxx ~]$ ssh -p 22222 root@10.100.100.102 cat /etc/os-release
ID="balena-os"
NAME="balenaOS"
VERSION="2.36.0+rev2"
VERSION_ID="2.36.0"
PRETTY_NAME="balenaOS 2.36.0+rev2"
MACHINE="raspberrypi3"
VARIANT="Production"
VARIANT_ID="prod"
RESIN_BOARD_REV="af53c77"
META_RESIN_REV="2fc8e3c"
SLUG="raspberrypi3"
[andrei@fluxx ~]$ ssh -p 22222 root@10.100.100.102 ping fluxx.local
PING fluxx.local (10.100.100.101): 56 data bytes
64 bytes from 10.100.100.101: seq=0 ttl=64 time=4.437 ms
64 bytes from 10.100.100.101: seq=1 ttl=64 time=3.876 ms
```

I suspect this is some kind of a racing issue because one an Intel NUC image I was able for a short period of time to replicate it. Do you have any additional information that might help in replicating it?

zrzka, 2019-05-28:
@agherzan I just flashed Fin & RPi3 with 2.36.0+rev2 and both devices area experiencing this issue. Is there anything I can help with? Like logs, ...? Or you can use my devices ...

* [My NUC dashboard link](https://dashboard.balena-cloud.com/devices/639c27b3e1def823daca60b0649d330c/summary)
* SSH to NUC & `ssh -p 22222 root@192.168.1.70` to access Fin
* SSH to NUC & `ssh -p 22222 root@192.168.1.71` to access RPi3

... I'll keep them running. Both are in a state where I flashed them & boot them. Nothing else. Both are connected via ethernet (no WiFi).

agherzan, 2019-05-28:
os-config fix: https://github.com/balena-os/os-config/pull/23
related fix in busybox: https://github.com/balena-os/meta-balena/pull/1534

commit 8d44decb (2019-05-28): os-config: Update to v1.1.1 to fix mDNS

os-config has a dependency on reqwest which broke mdns on 9.6. The new
os-config updated this version to one that restored this functionality.

Partially fixes: #1531

Change-type: patch
PR #1535 (2019-05-28): os-config: Update to v1.1.1 to fix mDNS
os-config has a dependency on reqwest which broke mdns on 9.6. The new
os-config updated this version to one that restored this functionality.

Partially fixes: #1531

Change-type: patch

---
### Contributor checklist

- [x] Changes have been tested
- [x] `Change-type` present on at least one commit
- [x] `Signed-off-by` is present
- [x] The PR complies with the [Open Embedded Commit Patch Message Guidelines](http://www.openembedded.org/wiki/Commit_Patch_Message_Guidelines)

### Reviewer Guidelines
- When submitting a review, please pick:
  - '*Approve*' [... cut]