balena-os/meta-balena #1533: initramfs crashes when kernel cmdline contains quoted string

svenschwermer, 2019-05-28:
When I set a kernel command line parameter that contains a quoted string (like a dynamic debugging parameter) in u-boot like so
```
=> setenv bootargs console=ttymxc5,115200 root=PARTUUID=3bd84f36-02 rootfstype=ext4 shell debug 'dyndbg=\\"file drivers/usb/* +p\\"'
=> printenv bootargs
bootargs=console=ttymxc5,115200 root=PARTUUID=3bd84f36-02 rootfstype=ext4 shell debug dyndbg="file drivers/usb/* +p"
```
the kernel recognizes the command line correctly:
```
[    0.000000] Kernel command line: console=ttymxc5,115200 root=/dev/mmcblk0p2 rootwait rw rootfstype=ext4 shell debug dyndbg="file drivers/usb/* +p"
```

However, the initramfs (shell?) crashes. No debug messages are printed, so this must be happening right at the start of the init process. Perhaps the shell chokes on the quoted parameter?
```
/init: eval: line 1: syntax error: unterminated quoted string
[    2.368099] Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000200
[    2.368099]
[    2.377266] Rebooting in 1 seconds..
```

agherzan, 2019-05-28:
Sounds links a bug in initramfs framework cmdline parser. I'll mark this as a bug and we will update this thread as soon as we have anything.

acostach, 2022-01-19:
Needs to be tested again as the initramfs scripts have been reworked in newer BalenaOS releases.