balena-os/meta-balena #1541: Improvement needed: Fix secondary gpt location after image is written to internal device

acostach, 2019-06-03:
For boards that use gpt there is a primary gpt header after mbr and a secondary one at the end of the disk. 

In the image generated secondary gpt header will be at the end of the image, however for the internal memory this won't be the case since it is larger.

With sumo I'm currently not seeing any issue on tx2 as flasher script does not fail and I can fix this with fdisk from the board specific flashing procedure script, last thing in the resin-init-flasher.

`bash[667]: '/mnt/boottmp/config.json' -> '/tmp/internal_boot/config.json'
 bash[667]: [resin-init-flasher] INFO: Board specific flash procedure...
`

I believe flashing does not fail because there's no error code returned when calling partprobe /dev/$internal_dev right after writing image to emmc.

However with zlk/warrior branch, which is not yet merged, the flashing procedure fails:

`bash[813]: Error: The backup GPT table is not at the end of the disk, as it should be.  Fix, by moving the backup to the end (and removing the old backup)?
bash[813]: Warning: Not all of the space available to /dev/mmcblk0 appears to be used, you can fix the GPT to use all of the space (an extra 56360935 blocks) or continue with the current >
systemd[1]: resin-init-flasher.service: Main process exited, code=exited, status=1/FAILURE
systemd[1]: resin-init-flasher.service: Failed with result 'exit-code'.
systemd[1]: Failed to start Resin init flasher service.`

For the future we may need something like either another hook similar to the board specific script but called between writing the image and doing partprobe, or some other mechanism to fix the secondary gpt header location directly in meta-balena.

ZubairLK, 2019-06-07:
@acostach I think the actual issue is more subtle. What you describe above seems to be a symptom.

In the case of flasher images, the data partition should not have expanded in the first place. I suspect it is due to the issue that was fixed by https://github.com/balena-os/meta-balena/pull/1506

Because the data partition expanded, the flasher script hit this. With https://github.com/balena-os/meta-balena/pull/1506, this shouldn't happen. Can you test if you can reproduce this with the current warrior branch in meta-balena?

acostach, 2019-06-11:
@ZubairLK I guess your are right, after fixing the wifi issue I tried again with the development branch, which now has the zlk/warrior one merged into it, and do not experience this anymore. Unless there's something else we can close this issue. Thank you for looking into this.

ZubairLK, 2019-06-11:
@acostach no worries.

closed 2019-06-11 by ZubairLK